tracespace/tracespace #2: Style example

henrikekblad, 2015-09-23:
Hi and thanks for this excellent library. 

I've managed to generate SVGs from my gerber files. 
Do you by any chance have a nice css example to use?

I'm trying to build a forum plugin (forum.mysensors.org) so the community users will get an auto-generate a svg from their gerber files in forum posts. 

If WebGL/threejs works as expected. I hope to create a 3D viewer from the SVGs.

mcous, 2015-09-24:
Ah glad to hear you've managed to generate something! This might be somewhat incomplete, as I'm recreating this from copy/paste and memory rather than my testing files. (I'm travelling right now and they're on my other computer, which is a weird problem to have in 2015).

I usually drop a style tag directly inside the SVG to ensure that if I encode in base64 or put it in an img tag, the style is still there. Given a board id of "foo", it would go something like this (for a nice green board with gold pads):

``` html
<style type="text/css">
  // substrate color
  .foo_board-fr4 {
     color: dimgrey;
  }

  // unfinished copper color
  .foo_board-cu {
    color: lightgrey; 
  }
  
  // finished copper color
  .foo_board-cf {
    color: goldenrod;
  }

  // soldermask color
  .foo_board-sm {
    color: darkgreen; opacity: 0.75;
  }

  // silkscreen color
  .foo_board-ss {
    color: white;
  }

  // solderpaste color
  .foo_board-sp {
    color: silver;
  }

  // outline color (only used if board shape calculation fails and it falls back on using lines)
  .foo_board-out {
    color: black;
  }
</style>
```

Let me know if you get this working. I'll try to follow up when I'm back home in a few days. If it does work, a wiki entry or README PR would be lovely. Also I'd love to hear any 3D progress. Mayhew labs stopped working for my boards years ago and I've been pining for an easy 3D viewer since (and had no time to write my own).

henrikekblad, 2015-09-24:
Thanks,

Might be good with an option to specify external stylesheet when generating top/bottom. 

Doing it afterwards on the html page is a bit hackish
http://stackoverflow.com/questions/4906148/how-to-apply-a-style-to-an-embedded-svg

henrikekblad, 2015-09-24:
Hmm.. after some more fiddling around... I cannot seem get external stylesheet to work... 
Tried adding <?xml-stylesheet href="style.css" type="text/css"?> at top of svg and browser seems to load the styleshhet but does not apply styles...

The only way I can get colors so far is by embedding <style></style> tag directly in svg body..

mcous, 2015-09-24:
I like the idea of allowing the user to pass in style.

I only ever insert the style directly into the SVG, mostly because I find that to be the simplest solution to all the problems I have (i.e. it's the only way I've found to ensure consistent SVG styling no matter how the SVG is consumed). So if we were to pass in style, I'd probably do it with a colors object that allowed you to pass in RGBA colors for all the elements. Then the stackup builder would build a style tag inserted inside the SVG.

So, not that I can be of much help, but is your SVG inline an HTML document or a standalone file inserted with an img tag or something?

henrikekblad, 2015-09-24:
Passing colors would be awesome. 

In my test.html file I just did a simple object embedding like this:

```
<object width="400px" data="bottom.svg" type="image/svg+xml">
</object>
```

mcous, 2015-09-24:
I did some playing around and got external stylesheets to work, at least with the pcb-stackup test board. I made a gist with the (simple) index.html and style.css that I used, as well as the file I used to generate the stackups in Node. Let me know if this helps: https://gist.github.com/mcous/ef394980a9f0e5634aa4

henrikekblad, 2015-09-24:
Wow, how cool is this! Ended up embedding the svg directly in forum post. Styles will work fine directly in webpage. 

[image]

Some feedback.... There seem to be Serial.prints which is good when debugging but it will end up in the forum log so it might be good to have an option to turn it off. 

When running my test-files I got an error here in drill file from a example project [https://github.com/mgebundy/LilyPad-SR/tree/master/gerber]

/Users/hek/git/mysensors/NodeBB/public/uploads/files/1443124211082-lilypadsr.drl
no drill file zero suppression specified. assuming leading zero suppression (same as no zero suppression)
no drill file units specified; assuming 2:4 inches format
no units set; assuming inches
[Error: Error at line 17 - Cannot read property 'flashed' of undefined]
at module.exports (/Users/hek/git/nodebb-plugin-gerber3d/node_modules/gerber-to-svg/lib/gerber-to-svg.js:90:13)

I will catch error-layers and run the rest through pcbStackup. But just thought you should know.

mcous, 2015-09-25:
Nice! Yeah, if the SVG is inlined in the page, CSS should work as it normally does.

Serial.prints? A little too long in the Arduino world? 😀But yeah, the console.warn messages are on by default, which is a poor design decision on my part. That behavior will be changed in v1.0.0 of mcous/gerber-to-svg. In the mean time, you can specify an array for the warnings to be pushed to instead:

``` javascript
var warnings = []
var layerSvgString = gerberToSvg(someGerber, {warnArr: warnings})
// no console.warns, and warnings contains any warning messages generated
```

Can you open an issue at mcous/gerber-to-svg with that drill file attached? I can look into that bug there.

mcous, 2015-09-25:
I'm going to keep this issue open as a feature request, so I don't forget

mcous, 2015-09-25:
Oh wait, I see why you're getting that error. `LilyPadSR.drl` is not an NC drill file, it is an Eagle generated drill file generation summary. `LilyPadSR.TXT` is the NC drill file.

If you still feel like opening a bug in gerber-to-svg, go for it. A file that isn't a gerber should gracefully indicate that fact, not throw an error.

henrikekblad, 2015-09-25:
> Serial.prints? A little too long in the Arduino world?

Haha... Day job using node.js and night hacking on MySensors/Arduino makes my brain a bit  schizophrenic.

> In the mean time, you can specify an array for the warnings to be pushed to instead:

Good enough for me!

I'll open a bug on the other project.

henrikekblad, 2015-09-26:
How is the board outline used in the final SVG? Somehow it doesn't seem to respect the outline file.

[image]

Using the following files:
http://forum.mysensors.org/uploads/files/1442657325784-mysensors-board-v3.2.3.zip

On the http://svgerber.cousins.io/ I can see border as gray, but shouldn't this area be green as well?. 

[image]

This is how it looks like on dirtypcb.
http://dirtypcbs.com/view.php?share=11495&accesskey=9a4200ea5d04f2725b5bf0a93925ed8c

henrikekblad, 2015-09-26:
Have you researched supporting Eagle brd-files?

To create a three.js 3d object I would need to extract the following information from the gerber files. Could you give me a hint how to go about this? 

```
var board = {
  "xMax": 54.28,
  "yMax": 59.68,
  "xMin": 0,
  "yMin": 0,
  "width": 54.28,
  "height": 59.68,
  "outline": [    // Board outline 
      0,0,
      54.28,0,
      0,0,
      0,59.68,
      0,59.68,
      54.28,59.68,
      54.28,59.68,
      54.28,0
  ]};
var holes = [{"x":26.67,"y":11.43,"drill":"0.8128"},
             {"x":34.29,"y":11.43,"drill":"0.8128"},...];
```

mcous, 2015-09-27:
So this is what I see on svgerber (I had to manually set BoardOutline.GML to a board outline, and it doesn't look like pcb-stackup recognizes `.gml` as an outline layer, so that's a separate issue):

[image]

The width and height stuff is pretty easy: as long as the outline layer is applied properly, the width and height properties of the svg itself will be the actual size of the board. If the outline is not applied properly, the height and width will be set such that they contain all image elements from all the layers. So:

``` javascript
var stackup = pcbStackup(layers, 'my-board')
var top = stackup.top

// strings with units, e.g. "1.25in"
var topWidth = top.svg.width
var topHeight = top.svg.height
```

As for the board outline and drills hits, that's a little tricker and you're likely to have a lot more luck extracting that information from the layer objects themselves rather than the SVG stackup. In the outline layer, there will be a [`path` object](https://developer.mozilla.org/en-US/docs/Web/SVG/Element/path) from which you can try to get points and such, and in the drill layer there will be a series of `use` objects that will reference certain `circle` objects.

I have thought about supporting Eagle `brd` files, but doing so would require a lot of rethinking of the render architecture as it currently stands (some of which I've started doing). Also I hate Eagle, but that's an entirely different story 😐.

henrikekblad, 2015-09-27:
> Also I hate Eagle, but that's an entirely different story.

Haha.. yeah.. The hardware people of the MySensors community mainly uses KiCad.  But still.. many people still uses Eagle of old habit.

Struggled last night trying to convert SVG into threejs texture (on the client). The closest I've come so far is using a library called "canvg" converting SVG into a Canvas. The html-canvas can be used as a source for a Texture. But all transparency fell off during the process which makes it a no-go.  
Need to fiddle with it some more before I give up that track. 

The only server-side (node) SVG->PNG libraries I've found seems to use headless Chromium for converting them which seems a bit overkill.

mcous, 2015-11-20:
Hey @henrikekblad, this isn't related to this issue, but if you're still trying to build PCB stuff in webGL, I've broken out all the gerber parsing and shape calculation stuff from gerber-to-svg into [gerber-parser](http://github.com/mcous/gerber-parser) and [gerber-plotter](http://github.com/mcous/gerber-plotter). Instead of spitting out SVGs, gerber-plotter spits out neutral shape information, so you could (in theory) use it to draw straight to an HTML Canvas. I haven't dug into Canvas at all, so I don't know how well it translates, but I thought I'd give you a heads up.

Otherwise the style stuff for this issue will be resolved post-v1.0.0 release of gerber-to-svg

henrikekblad, 2015-11-20:
Excellent news! 

I have been preoccupied with other projects for a while but still have the gerber project in the back of my head. I'll probably get back to you when I start testing it.

Cheers

mcous, 2016-05-04:
v0.1.0 adds [a color option](https://github.com/tracespace/pcb-stackup#color) where one can specify all layer colors. Default is:

``` javascript
{
  fr4: '#666',
  cu: '#ccc',
  cf: '#c93',
  sm: 'rgba(0, 66, 0, 0.75)',
  ss: '#fff',
  sp: '#999',
  out: '#000'
}
```

closed 2016-05-04 by mcous

shawon1220, 2022-08-23:
Hello @mcous . I am using webgl to load the glb 3d model to the pcb board, but the coordinate file provided by the user, I don't know how to convert the screen coordinates where the upper bit number of top.svg is located, please help, thank you.

The format of the coordinate file provided by the user is as follows:

Designator  	Layer	      Center-X(mm)	    Center-Y(mm)	Rotation
C1                   TopLayer	       97.5799	            140.644	        0
C2                   TopLayer	       97.5041	             147.1441             180
...