tracespace/tracespace #3: Outline doesn't apply

NoHomey, 2015-11-12:
Hellow @mcous  you probably remeber me I opened a issue on <gerber-parser>. It smees like the outline doesn't apply :(
 All my gerber files are with some deprecated futures of the current standart so at first I though that the outline is missing cuz of that but than I used the files from svgerber and the result is the same ...
[image]

This is the script that I use to generate the top and bottom files (I's almost copy-paste from the one from README but still ... )

```
var fs = require('fs')
var gerberToSvg = require('gerber-to-svg')
var pcbStackup = require('pcb-stackup')
var idLayer = require('pcb-stackup/lib/layer-types').identify
var walk = require('walk')
var path = require('path')
var walker = walk.walk(path.join(__dirname, 'samples'))
var gerberPaths = []
var layers = []
var files = []

walker.on('file', add)
walker.on('end', pcb)

function add(r, f, n) {
    var i = r + '/' + f.name
    files.push(i)
    n()
}

function pcb() {

    files.forEach(parse)

    function parse(filename) {
        var gerberString = fs.readFileSync(filename, 'utf-8')
        var layerType = idLayer(filename)
        var options = {
            object: true,
            drill: (layerType === 'drl')
        }
        var svgObj = gerberToSvg(gerberString, options)
        var p = {
            type: layerType,
            svg: svgObj
        }
        layers.push(p)
    }

    var stackup = pcbStackup(layers, 'mb')
    var top = __dirname + '/top.svg'
    var bot = __dirname + '/bottom.svg'
    var topP = gerberToSvg(stackup.top)
    var botP = gerberToSvg(stackup.bottom)

    fs.unlinkSync(top)
    fs.unlinkSync(bot)
    fs.writeFileSync(top, topP)
    fs.writeFileSync(bot, botP)
}
```

NoHomey, 2015-11-12:
I'm using the green board with gold pads style, you had recommended in #2

mcous, 2015-11-12:
Damn. Can you do me a favor and send me (link or email) the SVGs you generated? I'm running your code and you're right: pcb-stackup is spitting out some pretty wildly wrong results. It seems, at the very least, there's a serious issue going on with the [board shape module](https://github.com/tracespace/pcb-stackup/tree/master/src/board-shape) when used with the clockblock files.

NoHomey, 2015-11-12:
Ofcourse, you should have mail from ivo_stratev@abv.bg 
Ohh if you mean the wierd circle parts than all I can say is that when I include the produced SVGs wiith angular they magically disappears but when opened as single files in browser they appear, spend alot of time in searching  and I have no answer yet.

mcous, 2015-11-12:
The "weird circle parts" are caused by a bug in the board shape module. The board shape module is supposed to take all the paths in the outline layer and rearrange them so that they create a filled in shape. However, it seems to be taking straight line segments from the original path and re-drawing them as arcs.

This bug could be what's causing your problems. I'll have to dig into the code [here](https://github.com/tracespace/pcb-stackup/tree/master/src/board-shape) to see why this is happening.

mcous, 2015-11-12:
Looking at the generated SVGs, there appears to be some bigger problems going on. I'm going to need some time to figure this one out.

NoHomey, 2015-11-12:
Yea sure take all the time you need. And if there is anything I can help with, I'll be more than glad to do so.

mcous, 2015-11-12:
Ok I just pushed `v0.0.3` which resolves the issue with incorrectly placing arcs in the outline path. The svgerber sample files now work.

Hopefully this fixes the issues with your files. If not, check to make sure that `layerType` is getting set properly for your outline file and that your outline file doesn't draw an outline with small gaps.

NoHomey, 2015-11-13:
Thanks man. I have one question shouldn't pcb-stackup generate board outline layer just like svgerber does show a separate svg containing only the board outline ? Because I have a web project where I need to do some svg transformations and one of them is making the board outline more visible (And I was thinking that I can do that with jQuery by selecting the element with class id_board-out and changing the stroke and stroke-width but it seems like that class is missing from the generated files)

NoHomey, 2015-11-13:
Thanks man it worked.
I saw that id_board-out class is conditionally added in _stack-layers.coffee if the layer is found. So how can I change the style of the outline ?
Because I have a project where I need to change board outline style.

mcous, 2015-11-13:
The outline is only available to be styled if it does not create a fully closed shape. Then, it will just be drawn on, similar to the silkscreen. However, if the outline is a fully closed path, the stroke is removed and the path is filled in and used to [mask](https://developer.mozilla.org/en-US/docs/Web/SVG/Tutorial/Clipping_and_masking) the rest of the board. In this case, it has no style, because it's not a part of the stackup, it's just defining the overall shape.

I'm going to close this issue because the bug it was related to is fixed, but if you still have questions feel free to open up another issue with your questions or send me an email.

mcous, 2015-11-13:
If the outline is not fully closed, and it is drawn on, then setting `color` for `.id_board-out` should change the color of the stroke.

closed 2015-11-13 by mcous