tracespace/tracespace #4: End of file encountered before required M02 command

kasbah, 2015-11-22:
I trying to switch over from my hacked together [svgerber-board-builder](https://github.com/kasbah/svgerber-board-builder) but having some issues parsing the same gerber files.

[Here is a test I put together](https://github.com/kasbah/pcb-stackup-test) comparing `stackup` to `builder`. You can run the it with `npm run stackup` and `npm run builder`. Running `stackup` will give you 'end of file encountered before required M02 command' errors while `builder` will silently succeed.

mcous, 2015-11-22:
The reason you're seeing these errors is because gerber-to-svg is being sent files that aren't gerber files (before pcb-stackup is called). Builder uses two [try-catches](https://github.com/kasbah/svgerber-board-builder/blob/master/src/main.coffee#L33) to first try to process a given file as a Gerber, then as a drill, then ignore it if they both fail. With stackup, you'd have to do this yourself, possibly like:

``` coffee
for filename in process.argv[2..]
    gerberString = fs.readFileSync(filename, 'utf-8')
    layerType    = idLayer(filename)
    options      = {object: true, drill: (layerType == 'drl')}
    try
      svgObj       = gerberToSvg(gerberString, options)
      layers.push({type: layerType, svg: svgObj})
    catch error
      console.warn "could not parse #{filename} as #{layerType} because #{error.message}"
```

That being said, it definitely looks like identifyLayer says a `.drd` is a gerber outline (`out`) instead of a `drl`. So you were also getting errors because it was trying to process drill files as a gerbers. According to my unit tests, OrCad uses `.drd` for outlines, but seeing as I can't find anything online that actually confirms that, and I don't know anyone who uses OrCad, `.drd` should probably id as a drill.

I'd accept a PR to change that behavior, but I'm unlikely to do it myself as the next release of gerber-to-svg that will require a rethinking of stackup. But on the plus side, among several things, the new gerber-to-svg can autodetect the whether a file is a drill or a gerber.

kasbah, 2015-11-28:
Thanks for your help, looks like you opened #5 for the `.drd` files. Meanwhile I am happily running a try/catch like this

```coffeescript

for filename in process.argv[2..]
    gerberString = fs.readFileSync(filename, 'utf-8')
    layerType    = idLayer(filename)
    if layerType != 'drw' #drw is the default for any un-identifiable extensions
        try
            svgObj = gerberToSvg(gerberString, {object: true, drill: (layerType == 'drl')})
        catch e
            try
                svgObj = gerberToSvg(gerberString, {object: true, drill: true})
            catch
                console.warn "could not parse #{filename} as #{layerType} because #{e.message}"
                continue
            layerType = 'drl'
        layers.push({type: layerType, svg: svgObj})
```

closed 2015-11-28 by kasbah