tracespace/tracespace #6: Merge tools of all widths before finding outline?

kasbah, 2015-11-28:
This is what I did for [svgerber#23](https://github.com/mcous/svgerber/pull/23). Our test case for that was this MCHCK board, rendered using pcb-stackup below (with corners not cut off properly).

[image]

I know that I need to modify [src/board-shape/index.coffee#L13](https://github.com/tracespace/pcb-stackup/blob/master/src/board-shape/index.coffee#L13) and change a few tests as well so I wanted to ask if you think it is the right solution.

As an aside, the tests seem to rely on the input paths array mutating although this functionality isn't in actual use (is that right -- seems a bit odd)?

mcous, 2015-11-30:
I think the functionality is useful, but I also think that an outline file with an outline drawn with several different tools is a bad Gerber file. Therefore I think `pcbStackup` should take an options object where `mergeOutlineTools` is a flag that can be set. It should also have an `id` key that replaces the existing id argument and allow the options object to just be an id string:

``` javascript
// these two things are the same
var stackup = pcbStackup(gerbers, 'board-id')
var alsoStackup = pcbStackup(gerbers, {id: 'board-id'})
```

In terms of mutating the input paths array: it does, in fact, mutate it. This was a stupid idea and I don't know why I wrote it this way. Feel free to change that if you decide to tackle this.

kasbah, 2015-11-30:
Ace, I'll see when I find time to do this.

mcous, 2016-05-04:
Implemented in v0.1.0 of pcb-stackup as the `plotAsOutline` option in `gerber-to-svg@1.0.0` / `gerber-plotter`

closed 2016-05-04 by mcous