tracespace/tracespace #7: Fill gaps in outlines

kasbah, 2015-11-28:
As discussed on [svgerber#20](https://github.com/mcous/svgerber/issues/20). A good example are the Bus Pirate v3.6 gerbers (rendered by pcb-stackup below).

[image]

Svgerber would render this like so, with the outline not working properly. Not sure why it's different, maybe there is another issue.

[image]

Let me know if you still think repairing small gaps is the right approach, anyway.

mcous, 2015-11-30:
I just looked a more closely at the Bus Pirate outline you linked to in mcous/svgerber#20, and it has bigger problems than the small gaps (which it also has). The outline isn't one line; it's several lines drawn over each other in the same place. Dealing with this would be complicated™ (checking for duplicate segments might work), but this is certainly a poorly drawn design / Gerber file. At what point do we stop trying to accommodate for mistakes?

pcb-stackup tries to be more lenient than svgerber, so it will allow any path in the outline that seems manifold to be used as a mask (svgerber needs _every_ path in the outline to be manifold). Unfortunately, because the path goes back over itself with the bus pirate, it appears manifold even though it is messed up, which is why svgerber gives up on the mask but pcb-stackup does not.

I don't think this is a good case for adding gap fixing, but it is a good case for adding an option to force the board outline to be added as a drawing rather than a mask.

mcous, 2016-05-03:
Minor progress!

[image]

kasbah, 2016-05-03:
Ha, love it! Wouldn't be happy if I got that back from a fab :)

mcous, 2016-05-03:
Implemented gap-filling and duplicate segment detection, got this:

[image]

Trying to release all this tonight!

kasbah, 2016-05-04:
Wowza! Good work! Looks great!

mcous, 2016-05-04:
Implemented in v0.1.0 by using the `plotAsOutline` option in gerber-to-svg@1.0.0 / gerber-plotter

closed 2016-05-04 by mcous

kasbah, 2016-05-06:
I can't get the same result over on [kasbah/pcb-stackup-builder](https://github.com/kasbah/pcb-stackup-builder) (see [top.svg](https://github.com/kasbah/pcb-stackup-builder/blob/master/top.svg)).

What am I doing wrong?

mcous, 2016-05-06:
Looks like my example in the README doesn't use the outline as a mask by default. When you call `pcbStackup`, do this:

``` js
var stackupOptions = {id: 'my-board', maskWithOutline: true}
var stackup = pcbStackup(layers.filter(Boolean), stackupOptions)
```

The example in [the examples directory](https://github.com/tracespace/pcb-stackup/tree/master/example)  shows the use of this option.

Also the drill file needs to be manually set to leading zero suppression (aka keep trailing zeros).