tracespace/tracespace #9: Drills not rendered properly on FM_Transmitter board

kasbah, 2016-04-02:
Can't quite figure out why the drills aren't working properly with [this one](https://github.com/kasbah/pcb-stackup-test/tree/master/gerbers/FM_Transmitter).

Output: https://github.com/kasbah/pcb-stackup-test/blob/master/output/FM_Transmitter-top.svg

mcous, 2016-05-02:
It looks like the drill file uses 2:5 precision (default for drill files in inches is 2:4) and keep trailing zero suppression (drill file default is to keep leading zeros). I added

```
;FILE_FORMAT=2:5
INCH,TZ
```

to get the file to render properly (using gerber-to-svg@1.0.0 last night)

mcous, 2016-05-04:
Closing because it's a drill file issue and not a library issue

closed 2016-05-04 by mcous