tracespace/tracespace #11: Provide simpler API?

kasbah, 2016-05-06:
I am struggling a bit with moving from 0.0.4 to 0.1.0 in [monostable/kitnic](https://github.com/monostable/kitnic). The last API was quite low level, and this one is too so there are lots of details to work out.

It would be nice if pcb-stackup had a higher level API and I was thinking about implementing it. This steals ideas from gerber-to-svg and provides a stream but also a callback API. `pcbStackup` would use whats-that-gerber if no `layerType` is provided but the user could also specify a `layerType` themselves. Using `gerberToSvg` would not be needed.

I have outlined what I am thinking so far in Typescript below just so we can easier talk about the types. I would still write it in ES5. The `?` indicates an optional type that can remain `undefined`.

```typescript

function pcbStackup (files : File[], callback?) : StackupStream;

interface File {
    data       : string | Buffer | stream.Readable;
    filename   : string;
    layerType? : string;
}

function callback (error : string, stackup : Stackup) : void;

interface Stackup {
    top    : string;
    bottom : string;
}

interface StackupStream {
    top    : stream.Readable;
    bottom : stream.Readable;
}

```

What do you think?

mcous, 2016-05-06:
I'd definitely be down for a simpler API, and seeing as you're actually using this module in real life, I appreciate your opinions on it. Here are some initial thoughts on your proposal in no particular order:

* Automatic filename -> type shouldn't be a problem (I think we already talked about this and I meant to do it? My bad)
* I'd rather not have a streaming output from this, since the stackup function is not asynchronously consuming the streams, it's synchronously consuming the [public properties of the finished converter streams](https://github.com/mcous/gerber-to-svg/blob/master/API.md#public-properties)
    * I'd totally be down for a callback + maybe a Promise interface, too
* We need a way to pass options to gerber-to-svg, as well as a way to directly get at the output of gerber-to-svg
* I'd also like to keep the ability to pass in gerber-to-svg converters directly
    * This is for quickly regenerating the stackup if one wants to switch out a single layer or change the colors or something
    * The stackup building is the quickest part of the process, so we don't want to do the work of converting the gerbers unnecessarily

So, maybe something along these lines? (Forgive me for any syntactical errors; I've never used Typescript so [I have no idea what I'm doing](http://giphy.com/gifs/spinning-i-have-no-idea-what-im-doing-helicopter-f1ohcPEHABwWY))

``` typescript
function pcbStackup (layers : Layer[],  options? | callback?, callback?) : Promise[Stackup];

interface Layer {
    gerber     : string | Buffer | stream.Readable | GerberConverter;
    filename   : string;
    layerType? : string;
    options?   : GerberConverterOptions // maybe we can lump filename and/or layerType into here?
}

function callback (error : Error, stackup : Stackup) : void;

interface Stackup {
    top    : string;
    bottom : string;
    tcu    : GerberConverter,
    bcu    : GerberConverter,
    ...
}
```

kasbah, 2016-05-06:
Ok, great. I don't really know Typescript either but wanted some kind of standard for describing the types. I was mainly using [this](https://learnxinyminutes.com/docs/typescript/) for reference. It doesn't matter anyway, as long as we understand each other.

So `GerberConverter` is a [`stream.Readable`](https://nodejs.org/api/stream.html#stream_class_stream_readable) returned by gerber-to-svg right?

I can see the reasoning for the low-level api now, thanks for explaining where most of the computational time is spent. What if we call this re-usable object the cache and make it so it can be changed and passed back to `pcbStackup`

```typescript
interface Stackup {
    top    : string;
    bottom : string;
    cache  : Layer[];
}
```

Something I forgot to ask: how does all this node streaming stuff work in the browser?

mcous, 2016-05-08:
The Node streaming stuff works in the browser because, rather than depending on the built-in Node stream library, we depend on this [npm-land copy instead](https://www.npmjs.com/package/readable-stream).

By GerberConverter, yes, I meant the transform streams (that implement the stream.Readable interface) returned by gerber-to-svg.

I've been thinking about this over the past few days in terms of what I personally want/need as a consumer of this API, in order of importance:

1. Easy access to the top and bottom renders
2. Easy access to the individual layer streams so I can pipe the renders to places (e.g. S3) without having to wait for all of them (or the stackup) to finish
3. Easy access to render properties (e.g. size) of both the individual layers and the stackup outside of the SVG string
    1. This is already included in the properties of the gerber converter
    2. We should probably add some matching properties (width, height, and viewBox) to the Stackup return object

Other than (3ii), this currently can all be accomplished by wrapping the existing API in some extra logic. Anything you're looking for that's missing from this list?

As long as passing in GerberConverters is possible, I think I'm going to go back on what I said about wanting to return the individual layers in the Stackup object. If you pass in the GerberConverters directly, you have access to them, and if you don't need them, you don't need to worry.

So I think for Stackup we can go back to:

``` js
{
  top: string,
  bottom: string
}
```

I'm still waffling on what I think the Layer input should be:

``` js
{
  gerber: string | GerberConverter | stream.Readable // need to figure out how to differentiate these
  type: whatsThatGerber object | filename string
  options?: gerber-to-svg options if gerber is a string or filestream
}
```

kasbah, 2016-05-09:
I actually came around to the idea of having a cache. It lets pcb-stackup wrap the complexity of preparing all the data but also lets you cache and re-use it.

Right now my use-case is probably the simplest possible as I just have some gerbers and I want a render of the top and bottom. I don't actually need _any_ of the other features yet so that's how I approached the API. As I develop my site I will likely need to make use of the caching so I would appreciate an easy path to level up.

It's totally about wrapping the existing API into something more approachable. I have started working on this by re-purposing your example code in [kasbah/pcb-stackup-builder](https://github.com/kasbah/pcb-stackup-builder).

In regards to Layer type:

- The `GerberConverter` type needs some sort of type id so we can tell that it's a stream of converted Gerber.
- Providing a `whatsThatGerber object` instead of just a `string` of layer id would be annoying if you are not using whats-that-gerber. Not having to give a file name if you have already have a layer id would be really nice though. Maybe we could just use a type of  `{layerId: string, filename: string}` where you only have to provide one of the two.

mcous, 2016-05-09:
We could just duck-type the `GerberConverter` (e.g. if it's got a pipe method it's a stream, and if it has parser and plotter properties it's probably a gerber-to-svg stream).

The reason I suggested either a whats-that-gerber object or a filename is because it's easy to check if the parameter is a string or an object, and act accordingly. And a whats-that-gerber object is just an object with an id key (name isn't used): `{id: 'tcu'}`, so actually using whats-that-gerber isn't necessary if you don't want to.

Also, just to make sure we're on the same page, are we talking about modifying the API of this library or about creating a new package that wraps this library?

mcous, 2016-05-09:
Oh, and for reference, I was messing around this weekend and this was the simplest render board function I could come up with:

``` js
const gerberToSvg = require('gerber-to-svg')
const whatsThatGerber = require('whats-that-gerber')
const pcbStackup = require('pcb-stackup')
const shortId = require('shortId')

renderBoard: function(gerbers, options, done) {
    const layers = []
    const stackupOptions = Object.assign({
      id: shortId.generate()
    }, options)
    let layerCount = gerbers.length

    const finishLayer = () => {
      if (--layerCount === 0) {
        const stackup = pcbStackup(layers, stackupOptions)
        done(null, stackup)
      }
    }

    gerbers.forEach((gerber) => {
      const type = whatsThatGerber(gerber.filename)
      const options = {
        id: shortId.generate(),
        plotAsOutline: type.id === 'out'
      }
      const converter = gerberToSvg(gerber, options, finishLayer)

      layers.push({type, converter})
    })
  }
}
```

Where gerbers is an array of objects that are readable streams or strings with a property `filename` (this is adapted from a Hapi route handler, hence that interface, but you get the idea)

kasbah, 2016-05-09:
Yes, I am suggesting it for inclusion in pcb-stackup and that `renderBoard` looks pretty close to what the function would be. You could probably write it much faster than I could but I don't mind doing it.

The `id` thing is just a bit awkward either way, I would pass in an object with fields for both just for symmetries sake but it's not hugely important. 

What's your final verdict on the cache thing?

mcous, 2016-05-09:
If you think it's going to be helpful down the road for you I say go for it. I would probably just call it `layers`, though. We can either stick the entire converter in the output, or we can just copy the needed properties (`defs`, `layer`, `viewBox`, `width`, `height`, and `units`). I'd be more down for the latter, because it'll be easier to stick that into some sort of database (e.g. redis) and it'll be easier to differentiate a cached object from a new GerberConverter stream.

Yeah, the whats-that-gerber API I did at like 3AM the other day, so any suggestions to decrease the awkwardness are appreciated 😃 . I'm also toying with the idea of putting the layer type in the gerber-to-svg converter somehow (because the latest Gerber spec allows the inclusion of meta-information like layer type).

I just gave you write access to this repo if you wanna start playing around in a branch!

mcous, 2016-05-12:
I've got API changes that I think will help this open at mcous/gerber-to-svg#29 and tracespace/whats-that-gerber#2

kasbah, 2016-05-12:
Yeah, I saw the progress on those. I am not sure when I'll have time to work on this again so if you do want to implement it yourself instead of waiting around for me I wouldn't be disappointed.

kasbah, 2016-06-04:
Should I write this in ES5 or Coffeescript?

mcous, 2016-06-04:
If it's going in this module, please keep it ES5. If it's a wrapper module,
go wild! I've personally moved away from coffeescript for various reasons,
so my general preference is ES5 (it also makes test / coverage tooling and
npm install from GitHub way easier)

kasbah, 2016-06-04:
Ah yes, of course, turns out I was looking at an older version of pcb-stackup code and somehow thought this wasn't ported to ES5 yet.

kasbah, 2016-06-04:
Ok, so `Promise` is actually ES6 which we defined as our return type above. What do you want me to do with that?

mcous, 2016-06-04:
Oh, at the time I was playing around with Promises via [Bluebird](http://bluebirdjs.com/docs/getting-started.html). Seeing as (at 52.7 K minified) Bluebird is relatively heavy, I think it's fine just to do a normal callback interface with a note in the docs saying that one [can promisify the function](http://bluebirdjs.com/docs/api/promise.promisify.html) if a Promise interface is needed.

closed 2016-07-15 by kasbah