tracespace/tracespace #13: CLI tool

kasbah, 2016-06-05:
So we can flesh out some ideas. 

Names? 
- `gerber2pcb`, 
- `gerberup` 
- `stackup`
- `pcb-stackup`

We definitely want to take in shell expansion:
```
$ stackup gerbers/*
```

Easy to specify soldermask color with the rest adapting (similar to how kitnic.yaml color is handled) would be nice:

```
stackup --color red gerbers/*
```

We can start to collect some thoughts here.

mcous, 2016-06-06:
Shell expansion is handled pretty gracefully by the shell itself. Try this out to see what I mean:

`$ node index.js ./*`

``` js
// index.js
console.log(process.argv.slice(2))
```

If we want to get fancier and support globstars (not enabled on every shell by default, for example default OSX shell), the [glob](https://github.com/isaacs/node-glob) module works pretty great.

In terms of colors, we could do one or all of:

* What you suggested, with a `--color` parameter that specifies soldermask color
    * We should also allow the `color` option in the lib API to be a string as well as an object for this same functionality outside of the CLI
* Have a series of options that can specify call colors, e.g. `--color-fr4, --color-cu, --color-ss`
* Allow the user to pass in a CSS file that will be used as the style node
    * This option would be difficult with the way the current stackup API handles colors

kasbah, 2016-06-06:
Yes, no, I meant we want to take in a list of files. It's kind of obvious, I know. 

The rest sounds good. Is there no easy way to copy in CSS to the existing SVG output?

mcous, 2016-06-08:
We'd have to build an addition to the API to allow for a CSS string as the innerText of the style node. Not sure if its really worth it (at least for the first version) as long as the primary way of specifying colors is easy and expressive.

I'm partial to `gerber2pcb` and `pcb-stackup` for the name. I think pcb-stackup gets the advantage just because it's the name of the package and that, you know, makes sense.

mcous, 2018-07-03:
I'm planning on throwing this in the `4.0.0` release. I've got a little proof of concept going for a new module: `@tracespace/cli` that will function as both a replacement to `gerber-to-svg`'s CLI and as a CLI for `pcb-stackup`

```shell
# TODO
npm i -g @tracespace/cli

# render all layers and top.svg + bottom.svg to `board-renders`
tracespace board-gerbers/* --out=board-renders
```

commit 6c7c7999 (2018-07-16): feat(cli): Add @tracespace/cli package

Closes #13
commit a7704f80 (2018-07-17): feat(cli): Add @tracespace/cli package

Closes #13
commit 486d6fc8 (2018-07-17): feat(cli): Add @tracespace/cli package

Closes #13
closed 2018-07-17 by mcous