tracespace/tracespace #14: Example renders in the README

mcous, 2016-06-14:
[image] [image]

mcous, 2016-06-14:
Put the top render into the README with 781b6da

mcous, 2016-07-26:
[Some hackers are giving us crap](https://news.ycombinator.com/item?id=12157502) because the README image is a PNG file (hello Hacker News!).

Got a point, though, it's kinda stupid. Gonna push these example renders to gh-pages and reference them from there (image tag with SVG as src should work)

closed 2016-07-26 by mcous

kasbah, 2016-07-26:
Oh, that's where the traffic was coming from. Should we add this one? [image] 

The amazing thing about that one is that it's self inclusive!

mcous, 2016-07-26:
You have no idea how much I like that

commit c5057614 (2016-07-26): Use SVG renders in README and ADD MORE BADGES (fixes #14)
commit 53c84247 (2016-07-26): Merge pull request #30 from tracespace/use-vector-example

Use SVG renders in README and ADD MORE BADGES (fixes #14)