tracespace/tracespace #15: Drill still not found on fm-transmitter

kasbah, 2016-06-14:
I am using the test-fm-transmitter branch to figure out what is going on with this. I added zero:'L'  to the options like you suggested but it's still not working.

[image]

mcous, 2016-06-14:
Ah that was my bad, I forgot that I found two problems with fm-transmitter. It:

* Does not specify zero-suppression and uses leading-zero suppression by default (Excellon default is to keep leading zeros and suppress trailing zeros)
* Uses coordinates with 2 integer places and 5 decimal places, the Excellon spec says files in inches should use 2 integer places and 4 decimal places
  * NC drill files have no mechanism to express this format other than comments that some CAD programs add, and gerber-parser will catch them and use them
  * This drill file has no such comments

To get this to render properly, you need: `{zero: 'L', places: [2, 5]}`

kasbah, 2016-06-27:
Had messed up the option input, works with those options, thanks!

closed 2016-06-27 by kasbah

kasbah, 2016-06-27:
Is there any way we could detect a situation like this where additional options are required?

I just had a look at the viewBox without the right options and it's way out from the rest of them of course. Can we set some sensible thresholds there? I am not sure though why the `out`, `tss,` and `bcu` are so different from `tcu` and the `sm` layers and what that would mean for the thresholds. 

```
tcu [164.685, 65.51, 795.3149, 490.87]
drl [18425.19, 12025, 72600.81, 37052]
out [10, 10.51, 1000, 570]
drw [0, 0, 0, 0]
tss [10, 10.51, 1000, 570]
tss [10, 10.51, 1000, 570]
bcu [25, 25.51, 970, 540]
tsm [161.685, 62.51, 801.31489, 496.87]
bsm [161.685, 75.01009, 801.31489, 460.99982]
```

mcous, 2016-06-27:
Yeah I've been thinking about this as well. Given some arbitrary standard-deviation threshold, we should be able to throw a warning (not that we currently have a warning interface). We could even do some basic reasoning with what the problem is:

* Wrong units if viewbox is different by a factor of about 25.4,
* Wrong number of decimal places if different by a factor of about 10
* Wrong suppression or combination of problems if viewbox is just wildly different

On a side note, did you generate these Gerber's yourself? They've got the same problem as the Bus Pirate gerbers where the bottom silkscreen has a file ending that's usually used for top silkscreen (hence your two `tss` logs above).