tracespace/tracespace #16: Add option to pass in custom createElement?

kasbah, 2016-06-27:
With `createElement` now being an option in gerber-to-svg should we add an option to pass in a custom function in from here?

The usecase I am thinking of is creating badges for [kitnic.it](https://kitnic.it) where I would like to add background and text to the rendered SVGs. I guess what I really need is an object instead of string return and and then use xml-element-string after I have modified it. But I guess this is the same as having the `createElement` function be the [identity function](https://en.wikipedia.org/wiki/Identity_function).

mcous, 2016-06-27:
I'm pretty sure #12 already exposes this in a basic way by virtue of the fact that it allows one to pass in an options object for each layer. But if someone passes in `options.createElement` for the stackup, it should force all the `layer.options.createElement` to be the same.

For you particular use-case, you could also think about embedding the stackup SVG inside another SVG either directly or with a `<use>`

kasbah, 2016-07-11:
I was just looking at doing this but I was wondering: why isn't this done in pcb-stackup-core? Seems like there is no advantage to having the createElement functions be out of sync, or am I misunderstanding something?

mcous, 2016-07-11:
It has to be done here because by the time it gets to pcb-stackup-core, gerber-to-svg has already been called

commit 00ffc420 (2016-07-12): Force layer createElement option, fixes #16
closed 2016-07-15 by kasbah