tracespace/tracespace #17: Attempt different file on fail

kasbah, 2016-07-01:
Given a set of files, if the first identified file of that layerType actually fails to parse it would be good to fall back to another file with the same layerType.

mcous, 2016-07-01:
Failed parse should be pretty easy to detect by handling error or by looking for zero-size output (`result.width === 0 || result.height === 0`). However, unless the layer is a drill, having more than one layer of the same type seems like a problem, no?

I've been toying with the idea of throwing an error in stackup-core if more than one of certain layer types is passed in. e.g. there should never be more than one top copper layer, etc. Any thoughts?

kasbah, 2016-07-24:
Hmm, will think about this one and maybe try and do something before 2.0.0.

kasbah, 2016-07-24:
What's the advantage of handling this in core as opposed to here?

mcous, 2016-07-24:
Doesn't really make a difference to me. I can't remember the reason why I wanted to do that in core, so I'm assuming it wasn't important hah

kasbah, 2016-07-24:
Because we can probably still make the stackup with the layers we have I was just looking at creating warnings but there doesn't seem to be any convention. We could callback with an `Error` as err but still have a `stackup` as well. But there doesn't seem to be a normal way discern errors either though we could actually do something like:

```js
pcbStackup(layers, function(err, stackup) {
  if(err.severity > 3) {
    throw err
  }
  ...
})

mcous, 2016-07-24:
I like the idea of warnings for this. I think the best warning solution would be to have pcbStackup return an EventEmitter with `error`, `warning`, and `stackup` (or `data` or `done` or `whatever`) events (and obviously keep the callback). That way:

``` js
pcbStackup(layers, handleStackupDone)
  .on('warning', handleStackupWarning)
```

or even

``` js
pcbStackup(layers)
  .on('error', handleStackupError)
  .on('warning', handleStackupWarning)
  .on('stackup', handleStackup)
```

kasbah, 2016-07-24:
This one? https://github.com/Olical/EventEmitter

mcous, 2016-07-24:
Just meant [core EventEmitter](https://nodejs.org/dist/latest-v4.x/docs/api/events.html#events_class_eventemitter). It's easy enough for an end user to swap in a lightweight event-emitter module thru webpack / browserify configuration if needed for the browser.

kasbah, 2016-07-24:
I don't have any experience with swapping in Node core modules in practice. I see that in gerber-to-svg you depend on [readable-stream](https://www.npmjs.com/package/readable-stream). Wouldn't it make sense to depend on something like [event-emitter](https://www.npmjs.com/package/event-emitter) here?

mcous, 2016-07-24:
1. readable-stream is there to ensure the same latest Streams API is available across all platforms (see this old but I guess still relevant [article](https://r.va.gg/2014/06/why-i-dont-use-nodes-core-stream-module.html))
    * Without it, the gerber-* module test suites fail on 0.10
    * It's actually not for browser purposes, because:

2. By default, both webpack and browserify will automatically swap out core requires with user-land modules
    * For streams, both happen to use readable-stream
    * Here's the [full list of swaps that webpack uses](https://github.com/webpack/node-libs-browser)
    * You can also override manually; in browserify, for example, you can use [the browser field in package.json](https://github.com/substack/node-browserify#browser-field)

The core EventEmitter API hasn't changed (enough) since 0.10 for (1) to be a problem, and for browser environments, (2) is handled by either the build system or the user. Finally, readable-stream depends on core events, so it's already in the browser build.

TL;DR - it's fine to just `var EventEmitter = require('events').EventEmitter`