tracespace/tracespace #18: Is there a "hosted" version of this anywhere?

mithro, 2016-07-03:
If not, do you need hosting or something?

kasbah, 2016-07-03:
Also, it's a bit different but I am using this on https://kitnic.it.

mcous, 2016-07-06:
Hey @mithro, as @kasbah said, I'm working on a client-side version (to work like svgerber) and a server-side gerber-rendering-as-a-server. The client-side one is coming along (and will likely be GH Pages or S3), but the server-side one is little more than a proof of concept on my machine (and the reason the tracespace org exists).

I need more free time most of all! But some infrastructure would be good, as well. Haven't crunched any numbers, but (eventually) was hoping to throw it all up on AWS free-tier and see if I could ride that out. Did you have something specific in mind?

mcous, 2016-07-28:
We've added the [sites using this](https://github.com/tracespace/pcb-stackup#sites-using-this) section in the readme with some places this library is being used. Otherwise, there's some plans in store for tracespace. If you're interested in contributing in any way, please reach out!

closed 2016-07-28 by mcous