tracespace/tracespace #20: Problem finding outline of 8BitMixTape

kasbah, 2016-07-21:
The outline of [8BitMixtape/NextLevelEdition](https://github.com/8BitMixtape/NextLevelEdition) is not being rendered properly as noticed in  https://github.com/monostable/kitnic/pull/81.

I hade a bit of a play in [kitnic/NextLevelEdition](https://github.com/kitnic/NextLevelEdition) (see also the [re-plot branch](https://github.com/kitnic/NextLevelEdition/tree/re-plot) which gives even worse results) but didn't get anywhere. 

I have added this as a test-case in the [add-8bit-mixtape branch](https://github.com/tracespace/pcb-stackup/tree/add-8bit-mixtape) with a similar looking result. 

[image]

mcous, 2016-07-22:
The outline layer has several "large" gaps at many of the corners (especially the corners between lines and arcs). I say large, because (as mentioned over at mcous/gerber-plotter#4), they are larger than `plotAsOutline` can handle. If you look at only the outline layer, or turn off mask with outline, you should be able to see the gaps in the drawn lines.

Manually bumping the gap size in the path drawing module of `gerber-plotter` was able to render something, so I should be able to whip up a quick addition to that configuration option via `gerber-to-svg`. That being said, the larger the fillable gap size, the more visual quirks that get introduced.

kasbah, 2016-07-22:
Sounds reasonable, hopefully we can fix this one on the board itself but the option will still be really useful I think. 

Naive question: why does there need to be a threshold and what kind of quirks are you expecting? Isn't it a matter of finding the neirest neighbour if there is no exact joining point? Or is it because a line could join up with multiple lines?

As an aside, the silkscreen on this is styled white but appears grey, presumably because it is made up of lots of lines. I think on an actual PCB it would look white as well, any way we could match that? I am thinking maybe setting a minimum thickness for silkscreen lines would fix it? Is that right?

mcous, 2016-07-22:
Yeah, fixing it in the board is preferable, but a few extra control points never hurt anyone (that statement is patently false but whatever). It's been a hell of a week at work, but I should actually get to it tonight.

The threshold is there to prevent legitimate small details from being recognized as a gap and getting wiped out. A nearest neighbor search is certainly possible, but would still require a threshold (e.g. two points an inch apart, even if they are nearest neighbors, are not the same point), and would require an amount of work. I'm not sure the value is there. It would also (probably) slow things down.

Yeah, I'm not sure what we can do about the grey. It looks better if you zoom in. Unfortunately, even though Gerber files can specify fill regions, in every Eagle board I've seen, it creates regions with strokes instead. I'm not sure if there's a setting in Eagle to use regions instead, but that would be the best option. Your idea makes sense: we could add a `plotAsSilkscreen` option to the plotter that would thicken traces ever so slightly. I'm a little wary of that, though (and I'm not sure how much work it would take).

mcous, 2016-07-22:
Oh man, I just looked into it, and I was completely wrong about how it was constructing that fill. It's using a series of pads! I'm not sure there's anything we could do to make that look better, given the way the library works. It's also hell for Chrome to render, because each pad is it's own <use> element (as opposed to a series of strokes, which would all be in the same <path> element. Tweaking the Gerber output somehow may be the only way to go on this one.

I hate Eagle

mcous, 2016-07-23:
The latest version of gerber-to-svg (2.1.0) has the ability to set the gap option

kasbah, 2016-07-23:
What do you think about adding this as pcb-stackup option since setting the threshold is a bit awkward?

Something like

```js
pcbStackup(layers, {outlineGapFill: 0.011}, function(err, stackup) {
...
})
```
instead of:

```js
var idLayer = require('whats-that-gerber')

layers = layers.map(function(layer) {
  var id = idLayer(layer.filename)
  if (id === 'out') {
    layer.options.plotAsOutline = 0.011
  }
})
```

mcous, 2016-07-23:
I'm cool with it

kasbah, 2016-07-24:
Fixed in gerber-to-svg and pulled in here with a1c27507d6b8454a4bdbf959248c0645e189fe2e

closed 2016-07-24 by kasbah