tracespace/tracespace #24: 8-bit mixtape integration board needs drill file suppression set to leading

mcous, 2016-07-23:
Drill hits are missing from current image; the drill file uses unspecified leading zero suppression.

kasbah, 2016-07-23:
This seems to be a really common Eagle issue.

mcous, 2016-07-23:
Yeah it does. I want to look into trying to autodetect it in the parser if it's missing. Something where it buffers a few (20 would be enough for this board) coordinates and checks for 0s at either end of the coordinate first before deciding to fallback

kasbah, 2016-07-23:
I tried to ninja edit before: 
>Btw. do you prefer to push to master for one commit fixes like this or still go through a PR?

mcous, 2016-07-23:
Hah good question. I've been trying harder recently to put everything in a PR to help with history tracking. But I don't always stick to it. A PR would be great, no PR isn't going to kill anyone

kasbah, 2016-07-23:
You mean the merge commit or the record on GitHub? On my own code I generally merge single commit branches as fast-forward, i.e. no merge commit. Sometimes I will just push to master instead of going through the whole merge process if there isn't a branch already. Happy to send PRs here anyway though.

commit 89fa9ff5 (2016-07-23): Update options for 8bit-mixtape, also fixes #24
closed 2016-07-23 by kasbah