tracespace/tracespace #29: Make compatible with Inkscape

kasbah, 2016-07-25:
Try and figure out why Inkscape doesn't display the SVGs. 

8bit-mixtape: 

[image]

mcous, 2016-07-25:
I wonder if this is related to mcous/svgerber#26 (Also wow it's been nearly a year since that PR and I completely lost track of it...)

It looks like the solution there was to wrap the children of <mask> elements in a <g>; I wonder if the same thing would help here

kasbah, 2016-07-25:
I think you accidentally a `<g>`. Oh man, I totally forgot I found some workarounds for this in svgerber. Looking at the output SVG I can't see an obvious place for the groups though.

kasbah, 2016-07-25:
Tried this but didn't work
```diff
--- 8bit-mixtape_top.svg	2016-07-25 19:32:17.372072303 +0100
+++ 8bit-mixtape_top_grouped.svg	2016-07-25 19:33:31.912069666 +0100
@@ -23488,15 +23488,21 @@
       <use xlink:href="#S1h34QA7O_pad-8" x="1381.3" y="1386.5"/>
     </g>
     <mask id="Hyn47AQ__top_mech-mask" fill="#000" stroke="#000">
-      <use xlink:href="#Hyn47AQ__top_out"/>
-      <use xlink:href="#Hyn47AQ__top_drl1"/>
+      <g>
+        <use xlink:href="#Hyn47AQ__top_out"/>
+        <use xlink:href="#Hyn47AQ__top_drl1"/>
+      </g>
     </mask>
     <mask id="Hyn47AQ__top_cf-mask" fill="#fff" stroke="#fff">
-      <use xlink:href="#Hyn47AQ__top_sm"/>
+      <g>
+        <use xlink:href="#Hyn47AQ__top_sm"/>
+      </g>
     </mask>
     <mask id="Hyn47AQ__top_sm-mask" fill="#000" stroke="#000">
-      <rect x="168.2" y="102.3" width="3799.7" height="1540.1" fill="#fff"/>
-      <use xlink:href="#Hyn47AQ__top_sm"/>
+      <g>
+        <rect x="168.2" y="102.3" width="3799.7" height="1540.1" fill="#fff"/>
+        <use xlink:href="#Hyn47AQ__top_sm"/>
+      </g>
     </mask>
   </defs>
   <g transform="translate(0,1744.6999999999998) scale(1,-1)">
```

mcous, 2016-07-25:
Oh hang on this might have something to do with a weird CSS thing I did to get the outline to work in the mech-mask

1. Does it render properly without `maskWithOutline`?
2. Maybe try pcb-stackup-core@2.0.0; It uses a clipPath for the outline instead of sticking it in the same mask as the drill files

kasbah, 2016-07-25:
What other code changes are required for the update?

mcous, 2016-07-25:
No API changes, just did a major bump in case clipPath instead of mask broke things

kasbah, 2016-07-25:
Upgrading core gives:

[image]

Turning off `maskWithOutline` gives:

[image]

kasbah, 2016-07-25:
Grouping masks as above gets us the drills

[image]

mcous, 2016-07-25:
Hey look, progress!

If I didn't know any better, it looks like the styles aren't being applied. Comparing to svgerber, pcb-stackup does two things differently:

1. pcb-stackup places the `<style>` node inside of `<defs>`
2. pcb-stackup does _not_ add `type="text/css"` to the style node

kasbah, 2016-12-29:
Did some more investigating. Removing the CDATA stuff fixes it.

[image]

Full diff:

```diff
--- before.svg	2016-12-29 22:01:40.414884129 +0000
+++ after.svg	2016-12-29 22:00:43.998569898 +0000
@@ -1,13 +1,13 @@
 <?xml version="1.0"?>
 <svg xmlns:xlink="http://www.w3.org/1999/xlink" xmlns="http://www.w3.org/2000/svg" id="HkqwHZQHg_top" version="1.1" stroke-linecap="round" stroke-linejoin="round" stroke-width="0" fill-rule="evenodd" clip-rule="evenodd" viewBox="168.2 102.3 3799.7 1540.1" width="3.7996999999999996in" height="1.5400999999999998in">
   <defs>
-    <style>/* <![CDATA[ */.HkqwHZQHg_fr4 {color: #666;}
+    <style>.HkqwHZQHg_fr4 {color: #666;}
 .HkqwHZQHg_cu {color: #ccc;}
 .HkqwHZQHg_cf {color: #c93;}
 .HkqwHZQHg_sm {color: rgba(00, 66, 00, 0.75);}
 .HkqwHZQHg_ss {color: #fff;}
 .HkqwHZQHg_sp {color: #999;}
-.HkqwHZQHg_out {color: #000;}/* ]]> */</style>
+.HkqwHZQHg_out {color: #000;}</style>
     <rect x="-45" y="-45" width="90" height="90" id="BJu5vr-QHx_pad-12"/>
     <polygon points="36.99992,15.32587 15.32587,36.99992 -15.32587,36.99992 -36.99992,15.32587 -36.99992,-15.32587 -15.32587,-36.99992 15.32587,-36.99992 36.99992,-15.32587" id="BJu5vr-QHx_pad-13"/>
     <rect x="-19.7" y="-21.65" width="39.4" height="43.3" id="BJu5vr-QHx_pad-14"/>
@@ -35784,15 +35784,21 @@
     </clipPath>
     <mask id="HkqwHZQHg_top_mech-mask" fill="#000" stroke="#000">
+      <g>
       <rect x="168.2" y="102.3" width="3799.7" height="1540.1" fill="#fff"/>
       <use xlink:href="#HkqwHZQHg_top_drl1"/>
+      </g>
     </mask>
     <mask id="HkqwHZQHg_top_cf-mask" fill="#fff" stroke="#fff">
+      <g>
       <use xlink:href="#HkqwHZQHg_top_sm"/>
+      </g>
     </mask>
     <mask id="HkqwHZQHg_top_sm-mask" fill="#000" stroke="#000">
+      <g>
       <rect x="168.2" y="102.3" width="3799.7" height="1540.1" fill="#fff"/>
       <use xlink:href="#HkqwHZQHg_top_sm"/>
+      </g>
     </mask>
   </defs>
   <g transform="translate(0,1744.6999999999998) scale(1,-1)">
```

How do you want to proceed on this? 

For me it would be useful as svg2png is letting me down (well, PhantomJS or QtWebkit or whatever is -- it's crashing on some SVGs) and using Inkscape without GUI seems like a solid alternative.

kasbah, 2016-12-30:
Ok, so it doesn't like `rgba` either.

```diff
--- before.svg	2016-12-30 00:25:30.285613610 +0000
+++ after.svg	2016-12-30 00:30:24.639262394 +0000
@@ -4,7 +4,7 @@
     <style>.H1RI-FGXHe_fr4 {fill: #666;}
 .H1RI-FGXHe_cu {color: #ccc;}
 .H1RI-FGXHe_cf {color: #c93;}
-.H1RI-FGXHe_sm {color: rgba(00, 66, 00, 0.75);}
+.H1RI-FGXHe_sm {color: #004200; opacity:0.75;}
 .H1RI-FGXHe_ss {color: #fff;}
 .H1RI-FGXHe_sp {color: #999;}
 .H1RI-FGXHe_out {color: #000;}</style>
```

[image]
 
(switched to arduino because the mixtape silkscreen is a pain and slows everything down)

- Solder finish isn't showing up
- Solder mask isn't masked in all the right places yet

mcous, 2016-12-30:
Interesting! The CDATA is there to say that the stuff inside the CDATA shouldn't be treated as XML (https://stackoverflow.com/questions/2784183/what-does-cdata-in-xml-mean) so we can use special characters (https://stackoverflow.com/questions/21758345/what-are-the-official-xml-reserved-characters). It looks like we don't actually use any of those special characters, so we can probably ditch the CDATA as long as we protect / escape whatever's input as the `color` options.

I don't see any appreciable difference `color: rgba(...` vs `color: ...; opacity: ...`. Perhaps we should parse the the strings in `color` and write them as a base format (e.g. color and opacity). This would also take care of our special characters problem from above.

As for the finish and the mask stuff, the SVG output relies pretty heavily on `<use>` tags, and nesting `<use>` tags (e.g. a use for all the individual soldermask "pads", and then a use for the whole soldermask layer) can cause all sorts of problems. I wonder if that's what we're encountering here with InkScape and svg2png... There's def some potential for reducing the number of `use`s we do in gerber-to-svg

kasbah, 2016-12-31:
Just using gerber-to-svg the solder mask renders fine in Inkscape. I guess this means something is going wrong with the masking of it?

[image]

kasbah, 2016-12-31:
[Removing fill attributes in the solder mask declaration](https://gist.github.com/kasbah/07a3b0ed7aae8a48f69399d7399ab2ad) gives us the surface mount pads. 

[image].

I noticed there are loads of different circles and polygons with the same IDs, e.g `ryygLWKGmrl_pad-18`. How does it even know which one you are talking about in the `use` tags?

kasbah, 2016-12-31:
Adding `fill=#000` to the soldermask paths seems like an improvement in Inkscape.

[image]

But seems worse in browsers.

[image]

kasbah, 2017-01-17:
Closed by updating core with #46

closed 2017-01-17 by kasbah