tracespace/tracespace #31: Can't get the browser test to pass any more

kasbah, 2016-07-26:
It's getting quite annoying, even when I can get them to pass I normally have to restart the build a few times. 

Currently pressing restart on https://travis-ci.org/tracespace/pcb-stackup/jobs/147398037

mcous, 2016-07-26:
Same thing happened to me yesterday. Used to be a transient thing with Localtunnel, but it's been happening almost constantly recently.

We can [configure zuul to use ngrok instead of localtunnel](https://github.com/defunctzombie/zuul/issues/251) according to this thread. There's anecdotal evidence this will help. We'd drop from a maximum concurrency of 5 (Sauce-limited) to 2 (ngrok limited), though, so tests will take longer.

Want to try switching them in a branch and see if that helps? Also, do you have an Open Sauce account? This is much nicer to debug running from your own machine instead of Travis

mcous, 2016-07-27:
Just made very little progress trying to port over to ngrok over at core (branch `build-updates`). Might try again tomorrow. Goes poorly both locally and in travis: https://travis-ci.org/tracespace/pcb-stackup-core/jobs/147637579

mcous, 2016-07-27:
Very worst case: I have a Digital Ocean instance that's doing nothing and I bet we could host our own localtunnel

commit fcd861be (2016-07-28): Switch zuul tunnels to ngrox (fixes #31)
closed 2016-07-28 by mcous