tracespace/tracespace #43: Provide easier way to use React and similar createElement functions

kasbah, 2016-12-26:
Mainly creating this so we don't have to hunt for it in the gitter chat later.

For React this interface function below will work. Should we provide this in pcb-stackup?
Or maybe lower down in gerber-to-svg? The lack of unique key properties on some tags is also an issue.

```js
var React = require('react')
var camelCase = require('lodash.camelcase')

function createElement(type, props, children) {
  Object.keys(props).forEach(key => {
    var newKey

    if (key === 'xmlns:xlink') {
      newKey = 'xmlnsXlink'
    }
    else if (key === 'xlink:href') {
      newKey = 'xlinkHref'
    }
    else if (key === 'class') {
      newKey = 'className'
    }
    else if (/-/.test(key)) {
      newKey = camelCase(key)
    }
    if (newKey && newKey !== key) {
      props[newKey] = props[key]
      delete props[key]
    }
  })
  return React.createElement(type, props, children)
}
```

kasbah, 2016-12-26:
One thing to consider is that people might like to override the styles if they are putting svg tags into their html with the createElement funciton. In my usage I actually also have

```js
function createElement(type, props, children) {
  if (type === 'style') {
    return
  }
...
```
so I can define the styles in an external stylesheet.

mcous, 2016-12-26:
If you want to specify that a stackup should use external styles, it probably makes more sense to specify that via the [stackup color option](https://github.com/tracespace/pcb-stackup-core#color) rather than the createElement function:

``` js
const stackup = pcbStackup(layers, {
  id: 'some-id',
  colors: {
    fr4: false,
    cu: false,
    cf: false,
    sm: false,
    ss: false,
    sp: false,
    out: false
  }
})
```

Each element can be individually set to a color or set to a falsey value to use an external style

mcous, 2016-12-26:
I was doing some searching and couldn't find anything, but we should see if there's an npm module out there already does this (renames the props, I mean).

I don't think we should build it in because I don't want a React dependency in here. If we can't find anything on npm, we can create another module like tracespace/xml-element-string. No idea what to call it, though

kasbah, 2016-12-26:
No, I didn't mean to depend on React. More something that could be used like:
```js
const createElement = renameElements(React.createElement)
```

I wanted to search as well but didn't know what to search for. We are translating from the SVG spec to a just a React convention they have adopted for JSX? Or is it called something else maybe and more libs use it? Isn't it used in the DOM, e.g. `const a = documentQuerySelector('a'); a.className='my-link'` ? We could make a small module if there isn't one and simply document the usage here.

mcous, 2016-12-27:
Oh right, that makes more sense. But yeah, depending on how universal it is, could be useful generally as a small module.

React seems to be renaming SVG properties using some internal spec: https://github.com/facebook/react/blob/master/src/renderers/dom/shared/SVGDOMPropertyConfig.js

kasbah, 2017-01-02:
It's not much code and doesn't do `class` or `xmlns:...` but it's the closest thing I could find: https://github.com/stevenvachon/camelcase-css

kasbah, 2017-01-25:
I encountered a strange issue with this on [submit.preview.kitnic.it](http://submit.preview.kitnic.it/submit) with Gerbers from [BQ25570_Harvester](https://kitnic.it/boards/github.com/kitnic-forks/BQ25570_Harvester). The top copper layer wouldn't show up when using pcb-stackup client side. Adding those gerbers as an integration test here works fine though.

I tracked it down to an empty group with transform being added when run in the browser and the custom `createElement`. Ignoring the group in the `createElement` function sorts it:

```js
if (props.transform && type === 'g' && /_cu$/.test(props.id)) {
  return
}
```

mcous, 2017-01-25:
I'm not sure I understand the problem as described, but not creating an element for a group seems like it may be papering over a larger problem. There's no reason browser output should be any different from server output

kasbah, 2017-01-25:
It's definitely papering over the issue, yes. :laughing: I wasn't suggesting it as a permanent solution.

What the exact cause is, I am not sure but this group element isn't present on the server output. To dig deeper we could check if `xml-element-string` is seeing this and ignoring it somehow. 

 The only other difference I noticed is the ordering of props on the `<svg` element. But that seems reasonable since objects are not ordered.

kasbah, 2017-01-25:
This is the element that is being added for the BQ25570_Harvester that is causing the top copper to not be visible: 

```svg
<g id="pcb-stackup_top_cu" transform="scale(25.4,25.4)"/>
```

mcous, 2017-01-25:
That's really weird. That's a unit change scaling but all these Gerbers (as far as I can tell) are in mm, so that scaling should never appear.

mcous, 2017-01-25:
So I'm testing out this board on tracespace viewer, and there's definitely a bug somewhere in the stack where empty layers don't have units and then pcb-stackup-core incorrectly scales them because an empty string does not match `mm`. So that's frustrating.

But I'm not seeing that happen with the top copper layer, only with a bunch of the drawing layers / the NPTH drill file.

kasbah, 2017-01-25:
I see what's causing the issue now! On the Kitnic submit thing I am trialling trying to find the gerbers automatically by using any matches from whats-that-gerber. `cad/BQ25570_Harvester.cmp` is matching as `tcu` (but is actually a KiCAD file) so that's why I am getting an empty top copper. Sorry for the confusion!

I suppose I have to be a bit more clever than just filtering out everything that matches as 'drw'.

mcous, 2019-02-24:
After going through #104, I think _for most use-cases_ the best way to use tracespace renders in React is with `dangerouslySetInnerHTML`. Given:

1. The SVG renders tend to have many nodes
2. The renders are static; once a file is rendered it's going to change rarely, if ever (except for a potential brief period where the user may need to figure out the proper render options)
3. tracespace renders are full, sanitized markup documents, in my mind similar to the output of a markdown renderer

I think the best move here is to update the docs to either remove references to (or note that while you can, you probably shouldn't use) React.createElement with the `createElement` function, and that it would require a wrapper to work properly.

kasbah, 2019-02-25:
Yeah, makes sense. Maybe we should actually remove `createElement` from pcb-stackup options (can still stay in core of course)?

PR #110 (2019-03-09): refactor(pcb-stackup): Remove special handling of options.createElement
Closes #43. As discussed in that thread, the `createElement` option was added to help create renders directly in the DOM or VDOM. Given the number of nodes a given board render has and how efficient browsers tend to be at parsing `innerHTML`, using `createElement` for the creating of (V)DOM nodes turned out to be not that useful.

As suggested by @kasbah, this PR removes special handling of `createElement` in `pcb-stackup` to reduce the surface area of the API. The option still remains in gerber-to-svg and pcb-stackup-core [... cut]
commit 7d7fbebd (2019-03-09): refactor(pcb-stackup): Remove special handling of options.createElement (#110)

BREAKING CHANGE: pcb-stackup no longer tries to help if you use
options.createElement. You can still use it, but you'll have to manually
align your gerber-to-svg options with pcb-stackup's options

Closes #43
closed 2019-03-09 by mcous