tracespace/tracespace #44: Altium Gerbers result in SVGs that crash svg2png and don't work in Inkscape

kasbah, 2016-12-29:
The plots of [these three Altium projects](https://github.com/pcbsz) seems to be making it impossible for me to convert the SVGs to PNG.

 I already [filed with svg2png](https://github.com/domenic/svg2png/issues/77) as they cause it to crash and it's reported further upstream with PhantomJS. A PhantomJS downgrade [makes for some crappy pngs of these three projects](http://png-preview.preview.kitnic.it). Even without svgo optimization and fixes from #29 applied to this it doesn't render properly in Inkscape.

[image]

In inkscape:

[image]

Any guess why these renderers don't like these SVGs?

mcous, 2016-12-30:
Bold. Do they render properly in the big browsers?

kasbah, 2016-12-30:
What do you mean by "bold"? Yeah they render properly in browsers. I wasn't going to open an issue until I thought inkscape was having problems with them too which turned out to be false.

mcous, 2016-12-30:
Haha sorry, "bold" as in a "bold move" on those gerber file's part. [Really didn't mean anything at all](https://www.youtube.com/watch?v=4Ru8DMW-grY).

Yeah, best guess is we're just overwhelming the renderer. Might need to look as simplifying the output of g2s as mentioned in #29

mcous, 2018-07-22:
Just spent some time on the issue tracker for svg2png and got caught up on the sunsetting of phantomjs. Given that this specific issue looks to be a bug in those tools, (that won't be fixed?) I'm gonna close this ticket.

Continuing to track simplifying the SVG output in tracespace/tracespace#80

closed 2018-07-22 by mcous