tracespace/tracespace #48: PCB Outline Not Correct

alexgoldstone, 2017-07-12:
Please see contents of [files.zip](https://github.com/tracespace/pcb-stackup/files/1140557/files.zip) attached.

**output.svg** was rendered by pcb-stackup.

**expected.pdf** was rendered automatically by the quotation tool at eurocircuits.com

The output from pcb-stackup is always a solid filled rectangle and never represents the correct outline of the PCB for any of my Gerber files (I have tried several).

These Gerber files were output by OrCAD.

The same Gerber files were provided to the online tool at eurocircuits.com which renders the PCB correctly so there seems to be enough data in the files.

I have also included the application code (using pcb-stackup) that generated the output.svg file.

Attached:
[files.zip](https://github.com/tracespace/pcb-stackup/files/1140557/files.zip)

kasbah, 2017-07-12:
Can't try it myself right now but have you tried increasing the outline gap fill, e.g. `{outlineGapFill: 0.011}` or even higher?

alexgoldstone, 2017-07-12:
I have used values 0.011, 0.1 and 1.0 but unfortunately there was no perceptible difference with regard to this issue.

mcous, 2017-07-12:
Hey @alexgoldstone, I don't see a board outline file in any of these gerbers. Unfortunately, this tool needs an outline gerber file to be able to properly display the board shape

mcous, 2017-07-12:
Unrelated: whats-that-getber is not properlying ID'ing `*.npt` as a drill file: tracespace/whats-that-gerber#14

alexgoldstone, 2017-07-12:
Hi @mcous, that is helpful, thank you. I will investigate generating an outline file.

Just to confirm, it wasn't an oversight, we have never generated a board outline file and yet the automated systems used by the PCB manufacturers seem to infer the outline just fine.

Do I just consider that a difference between pcb-stackup and other tools or could this become a feature request to do the same?

mcous, 2017-07-12:
> Do I just consider that a difference between pcb-stackup and other tools or could this become a feature request to do the same?

Inferring board outline would take a lot of work for this tool, given that it's mostly just a vector image conversion tool at its core (gerber to svg). The amount of math and image inference it does is minimal, and at the moment almost entirely limited to processing a board outline file in isolation. TL;DR - Outline inference is not a likely feature of pcb-stackup

> Just to confirm, it wasn't an oversight, we have never generated a board outline file and yet the automated systems used by the PCB manufacturers seem to infer the outline just fine.

Interesting! I am curious as to how they infer it. That being said, I've never heard of anyone not generating an outline file for production boards, as every fab I've ever used recommends or requires one. Eurocircuits, for example, definitely recommends a board outline layer (Rule 10 on http://www.eurocircuits.com/10-rules-for-better-data/, http://www.eurocircuits.com/PCB-design-guidelines/#mechanicallayer)

alexgoldstone, 2017-07-12:
[KB0018R00.zip](https://github.com/tracespace/pcb-stackup/files/1142174/KB0018R00.zip) is the exact archive sent to Euro Circuits which 'just works'. I have sent the same to other PCB manufacturers without issue.

The only difference between this archive and the previous example is the inclusion of a DRD file.

I removed the DRD file when using pcb-stackup because using the DRD file takes me even further away from the outlined PCB I am after. Please see [top.svg.zip](https://github.com/tracespace/pcb-stackup/files/1142224/top.svg.zip)

Should I have kept the DRD file and done something else to avoid the inclusion of the dimensions and drill chart? Or, is the DRD file not the outline file you are referring to?

Attachments:
[KB0018R00.zip](https://github.com/tracespace/pcb-stackup/files/1142174/KB0018R00.zip)
[top.svg.zip](https://github.com/tracespace/pcb-stackup/files/1142224/top.svg.zip)

kasbah, 2017-07-12:
The DRD layer doesn't look like it's meant to be machine parse-able but it
does have the outline on it. I suspect the fabs edit in an outline layer by
hand because it's quicker than asking you to provide one.

alexgoldstone, 2017-07-12:
Thank you. So, that explains how our boards are being manufactured but not how the automated tools are inferring the outline (if they aren't using the DRD file) as Euro Circuits' automated analysis tool generates a rendered image that looks correct.

mcous, 2017-07-12:
Just ran the files through http://viewer.tracespace.io, and the DRD file is definitely a Gerber file that contains the outline, which explains how the fab is able to manufacture the board (like @kasbah said, they're pulling the data from there rather than inferring anything). Unfortunately, it also contains a bunch of other information (and possibly a few minor errors) that make in unsuitable for the naive board shape interpretation we run in pcb-stackup

alexgoldstone, 2017-07-12:
Thanks @mcous.

I'll see if the person that generates the files has an option to output just an outline on its own.

Thank you for your guidance regarding this. It has been very helpful in enabling me to know what to ask for.

Euro Circuits and others must have a different process for generating their image renders because there's definitely no human intervention performed as part of their analysis and I get a board image from them (outlined as expected) a couple of minutes after upload.

mcous, 2017-07-12:
pcb-stackup uses a very naive method for determining board outline: it looks at the outline layer, tries to fill in any small gaps, and then applies it as an SVG clip-path. Like I said, more than anything this is an image conversion library.

In the case of this particular outline file, there are two things that confuse gerber-to-svg and pcb-stackup:

1. Two tools are used in the drawing layer: one to draw the outline and one to draw all the dimensions and drill callouts
    - pcb-stackup is not capable of knowing which is which
    - By default, it'll assume all data in the outline layer represents outline data
2. The top right arc of the outline isn't very well defined by the orcad output, and when gerber-to-svg does its arc math it determines that the arc, as described by the gerber data, is impossible
    - Gerber and SVG describe arcs in different ways, hence the necessity of math and conversion
    - Gerber files have a built in maximum tolerance, which leads to rounding errors, which leads to potential problems with arcs. Generally we try to be lenient, but that doesn't always work
    - https://github.com/mcous/gerber-plotter/issues/5

Eurocircuits probably does a more sophisticated analysis of the manufacturing files, hence the ability to pick out the outline and correct for gerber arc errors and has probably seen many more gerber files than I have 😄

alexgoldstone, 2017-07-13:
Hi @mcous

I will be grateful for some further assistance if you don't mind.

My colleague has generated the attached [DRD file](https://github.com/tracespace/pcb-stackup/files/1146635/Archive.zip)

This contains only the board outline.

_Just a side-note, he advised that he does not usually do this as the default from Orcad includes not only the outline but also the drill hole positions and any other manufacturing dimensions he wants to add and manufacturers accept it. It did not however seem to be a problem for him to output an outline only file._

The file he sent me had a .BRD extension. This didn't make any difference to the resulting svg output.

I renamed the file to make the extension .DRD and can now see the outline in the resulting svg file.

**Unfortunately, even with the outline added, the final image is not clipped to it.**

I loaded the new outline file into trace space viewer and (unlike the previous outline) it seems to be intact so would have expected it to work.

Any advice will be greatly appreciated.

[Archive.zip](https://github.com/tracespace/pcb-stackup/files/1146635/Archive.zip)

mcous, 2017-08-01:
Hey, sorry for the delay here! I was on vacation and away from GitHub.

Two things are causing problems here:

1. Your drill file has the extension `.DRD`, which unfortunately Eagle uses as a drill file format
    * pcb-stackup and tracespace viewer use [whats-that-gerber](https://github.com/tracespace/whats-that-gerber), which gives Eagle precedence because it's more common in the OSH community, so they think your outline file is actually a drill file
    * I changed the extension to `.FAB`, which is what I have on record as the filetype for Orcad outlines, to get the automatic filetype checking to work
    * Sort of tracking this at #49, but it wouldn't fix this particular issue

    [image]

2. The outline file has an useless Step-Repeat block that's confusing the outline mask (because of the naive gerber to SVG translation used by pcb-stackup)
    * The offending line is 19: `%SRX1Y1I0.0J0.0*%` which says
        * x-repeat: 1, x-offset: 0
        * y-repeat: 1, y-offset: 0
        * In other words, put the image in once where it already is... i.e. don't repeat it
    * I removed this line and the board render worked
    * Opened a ticket because this shouldn't mess things up: mcous/gerber-to-svg#38

[image]

kasbah, 2018-04-10:
FYI I tried both of the provided zips with OshPark, DirtyPCBs, Aisler.net and Gerblook.org and none of them could determine an outline for any of them.

mcous, 2018-07-23:
Issues with this board's renders are:

- Outline layer had other stuff drawn on it (cannot support)
- SR disable line modifies modifies render when it should noop, breaking board outlint (bug, #81)
- Arc processing is too strict (bug, #82)

Gonna close this ticket in favor of the individual bugs listed above

closed 2018-07-23 by mcous

mcous, 2019-03-24:
@alexgoldstone sorry about the super slow pace of development, but as of 4.0.2, tracespace is able to properly render the second set of gerbers (`Archive.zip` - with the outline-only DRD file). This report led us identify some annoying bugs, so thank you!