tracespace/tracespace #51: Output board dimensions in mm by default

kasbah, 2018-03-21:
Rather than the dimensions used in the Gerber files. The `outlineGapFill` option should also be in mm by default.

mcous, 2018-03-21:
Rather than switching the default, should this be an option?

ju5t, 2018-03-21:
An option sounds good too. It definitely helps if you know which dimensions to expect.

mcous, 2018-03-21:
> I think having it vary by Gerber file is never what you want

I kinda disagree. I think it would be confusing to upload a bunch of gerber files in inches and get back a render in millimeters (or vice versa; I would argue against a default of inches, too).

Alternatively, this is a web format, so maybe we do it in `px`? I think the "reference pixel" is 1/96th of an "inch" (https://www.w3.org/TR/CSS21/syndata.html#length-units)

kasbah, 2018-03-21:
I guess I don't understand your use-case. What units are used in the display is a UI issue. Most people don't care what units their gerber files are in but will care what units are used in a UI. 

Anyway, I am ok with it being an option too (`{forceUnits: 'mm' | 'in'}`) though I think it makes less sense. Not a big deal. 

Having it vary also makes `outlineGapFill` pretty unwieldy. Can we at least have that always in `mm`?

mcous, 2018-03-21:
Yeah `outlineGapFill` has always been kind of a kludge, so I'm not really invested.

As for the units thing, I'm mostly approaching it from the standpoint of:

1. Not introducing breaking changes when an option will do
2. The units are involved all the way down the stack
    a. A given layer is going to be in. or mm. depending on the Gerber
        - Gerber coordinates are effectively mirrored in SVG land because the whole conversion is (basically) just a mapping function
    b. A unit difference between a layer and the board adds a `transform` attribute to that layer in the stackup
    c. Defaulting the board to `mm` without also defaulting the entire conversion process to `mm` would mean using Gerbers in inches creates board renders that are just a little bit harder to display every time
        * (Aside: defaulting the whole stack to `mm` might be good idea, but is a much larger project) process

mcous, 2019-03-09:
In hindsight, (and based on debugging some outlines with gaps) I think the switch to mm for `outlineGapFill` may have been a mistake. From what I can tell, outline gaps are often a function of resolution loss during gerber output from CAD. The resolution loss is caused by the fixed-decimal coordinate format of the Gerber file, which quantizes the locations that a segment's endpoints can land on.

When `outlineGapFill` was "unit-less", it was (almost) functioning as an epsilon, at its best saying "if two outline points land on adjacent discrete coordinate spots, consider them at the same point." Specifying `outlineGapFill` as layer units was a confusing choice, though. Also, it wasn't actually functioning as an epsilon because its value wasn't tied to the coordinate format; its default value was simply selected to line up with the most common format.

Since the previous way of specify `outlineGapFill` was definitely more confusing than locking it to `mm`, I don't want to revert it for 4.x. For the 5.x line, though, I think the most proper thing to do will be to change the option to an epsilon that is tied to the coordinate format. For example, `1.5` would actually do the "merge adjacent points in the matrix" thing.