tracespace/tracespace #73: This drill file is looks bigger than expected

kasbah, 2018-07-05:
Can't figure out what's going on with this .drd file made by @sangavalerian and @rwb27 using Eagle I believe. 

https://github.com/rwb27/openflexure_nano_motor_controller/blob/18a06fb9a9e6a0a873b6234e204a5500d0b735d9/sangaboard_v0.2/sangaBoard-v0.2%20gerber/sangaboard_v02.drd

It renders much bigger than expected with both gerber-to-svg and gerbv. gerber-to-svg warns:

```
warning at line 15 - zero suppression missing; detected leading suppression
warning at line 13 - notation not set; using backup notation: A
```

I tried to add all the possible combinations of `--units` and `--zero` but nothing seems to make it better (only worse). Is this actually just plotted much bigger than expected for some reason?

kasbah, 2018-07-05:
Looks like the original was plotted with Eagle v9. I opened it with the Eagle v7.5 I have installed and it looks fine in the brd, if I replot, it has the same issue.

kasbah, 2018-07-05:
I found that setting this in gerbv puts it into the right place. I haven't figured out how to tell gerber-to-svg to use "5 digits" (I thought it was `-f'2,5'` ?) or embed that into the drd. 

[image]

kasbah, 2018-07-05:
Ah, looks like it's `-f'[2,5]'`. Now, I guess the only way to detect this would be to look at the accompanying .dri file?

mcous, 2019-02-24:
I think architecturally this speaks to a need to process all of a board's files together rather than separately, taking #77 to its logical conclusion. I don't think this is something that can happen in the 4.x line, so I'm gonna close this issue because we have the appropriate options to specify this override.

Will be giving this more thought as I also think about changes I want to make to the per-file conversion pipeline

closed 2019-02-24 by mcous