tracespace/tracespace #78: shortId.generate() occasionally returns invalid id

ldstein, 2018-07-18:
Hi,

I noticed PCB Stackup will intermittently output an invalid SVG. Depending on viewer, it will display either as a black or blank image.

It appears to caused by  `shortId.generate()` occasionally returning a string which starts with a numeric character. For example "0C3Vwf2nt". 

XML IDs aren't allowed to start with a numeric character, causing the SVG viewer to misbehave.

mcous, 2018-07-18:
Good catch! Fairly embarrassing oversight on my part. Related to tracespace/gerber-to-svg#41 (ID sanitization) and will fix them both up together

davidworkman9, 2018-10-12:
v4.0.0-next.13 still has this problem. It's caused by a period in the ID now instead of starting with a number.
[image]

mcous, 2018-10-12:
Oh shoot. Period in the ID is a valid XML ID but period inside a class name without escaping it is not. Sorry for missing that

closed 2018-10-13 by mcous

mcous, 2018-10-13:
Fix released to:

```
@tracespace/xml-id@4.0.0-next.14
gerber-to-svg@4.0.0-next.14
pcb-stackup-core@4.0.0-next.14
pcb-stackup@4.0.0-next.14
@tracespace/cli@4.0.0-next.14
```

commit 98488bb6 (2018-07-21): fix: Replace shortid with @tracespace/xml-id and sanitize IDs

Closes #78
PR #88 (2018-10-13): fix(xml-id): Restrict characters to those valid in CSS selectors
This PR removes dots (`.`) from the output of `@tracespace/xml-id` because they are not valid to use in CSS selectors, and we use these IDs verbatim in CSS selectors. Rather than try to keep dots in the output and sanitize any time we create a selector, it feels more straightforward to just remove them from the ID alphabet.

Fixes #78
commit 1fe779e4 (2018-10-13): fix(xml-id): Restrict characters to those valid in CSS selectors (#88)

Fixes #78
commit 2cda7606 (2018-11-02): fix: Replace shortid with @tracespace/xml-id and sanitize IDs (#79)

Closes #78