tracespace/tracespace #83: Operation with non-existent tool should no-op with a warning

mcous, 2018-07-22:
This ticket is taking over for mcous/gerber-plotter#14.

Although it should be exceedingly rare (as in, I've only seen it happen as a result of other bugs in the stack), the plotter crashes hard if it attempts to operate using a tool that isn't defined.

## expected behavior

If the plotter encounters an operation using a tool that isn't defined, it should emit an error and no-op.

## actual behavior

The plotter throws a TypeError that isn't `catch`able because it's happening in an async handler.

m59peacemaker, 2019-08-03:
I'm *very* new to working with gerber files. I ran into this error using `gerberToSvg` on this file:

```
%FSLAX24Y24*%
%MOIN*%
%SFA1.000B1.000*%

%MIA0B0*%
%IPPOS*%
%LNsquareCUTOUT*%
%LPD*%
G36*
X-10000Y-10000D02*
G01Y10000D01*
X10000*
Y-10000*
X-10000*
G37*
%LPC*%
G36*
X5000Y-5000D02*
G01Y0D01*
G75*
G03X54Y5000I-5000J0*
X-5000Y0I-54J-5000*
G01Y-5000*
X5000*
G37*
%LPD*%
M02*
```

I don't know what it all means yet, but I know that removing these two lines fixes it:

```
G03X54Y5000I-5000J0*
X-5000Y0I-54J-5000*
```

Further, I could simply replace the "J" character in those two lines and I don't get an error. I arbitrarily put a "D" in place of those and didn't get a crash.

I'm interested in making a PR for this, but I'd need some guidance to get started. I don't know if this Gerber is just bad and shouldn't be usable, it's bad but should be usable, or it's fine and there's a bug anyway.

mcous, 2019-08-04:
Hi @m59peacemaker, that looks like a separate but related bug, which could definitely be tackled in a fix PR for #83. The line that's throwing looks like it needs its if conditions switched around:

https://github.com/tracespace/tracespace/blob/885b1500a3db61cb493290b82175698d72066ddf/packages/gerber-plotter/lib/_operate.js#L384-L388

This particular Gerber exposes this bug because:

- There are no tools (which would define pad shapes or trace thicknesses) in this file, because it's only using region draws
    - We need to check if we're in region mode before accessing the (maybe undefined) tool
- This particular if block is only hit during arc draws (`G03` operations)
    - Replacing the `J` makes the command invalid, in which case it gets skipped altogether

To fix the original bug this ticket is referencing, we need another `if` at the very top of this `interpolate` function that checks `if (!region && !tool)`, logs a warning, and then returns an empty box. Basically what the other two "noop" ifs in this function are already doing.

Is that enough information to tackle a PR? Happy to give more context if needed. I'd also recommend checking out the README's [contributing section](https://github.com/tracespace/tracespace/blob/master/README.md#contributing)

m59peacemaker, 2019-08-06:
There are a lot of possible states in that function. I can't feel confident about touching it since I lack general knowledge about the topic and the library or how to write tests that will cover these possible branches.

Does this part have to be placed after the `mode === 'i'` condition?
https://github.com/tracespace/tracespace/blob/master/packages/gerber-plotter/lib/_operate.js#L384-L388

If it could go before, some of the complexity could be reduced by putting the warning conditions up front and having a single block for those conditions:

```js
  var warning = null

  if (!region) {
    if (!tool) {
      warning = 'tool is not defined'
    } else if (tool.trace.length === 0) {
      warning = 'tool ' + tool.code + ' is not strokable; ignoring interpolate'
    } else if (tool.trace.length !== 1) {
      warning = 'cannot draw an arc with a non-circular tool'
    }
  }

  if (warning) {
    plotter._warn(warning)
    return boundingBox.new()
  }

  if (mode === 'i') {
    // add a line to the path normally if region mode is on or the tool is a circle
    if (region || tool.trace.length === 1) {
      return drawLine(start, end, tool, region, pathGraph)
    }

    // else, the tool is a rectangle, which needs a special interpolation function
    return interpolateRect(start, end, tool, pathGraph, plotter)
  }

  return drawArc(etc)
```

Maybe if I had some clarity on the necessary conditions for these calls, I could refactor it in a pleasant way. I doubt this is accurate, but it's the best I could do as a starting place:

```js
  mode === 'i'
    && (region || tool.trace.length === 1)
  drawLine()

  mode === 'i'
    && !region
    && (tool.trace.length === 0 || tool.trace.length > 1)
  interpolateRect()

  mode !== 'i'
    && (region || tool.trace.length === 1)
  drawArc()
```

m59peacemaker, 2019-08-06:
An example of where my lack of knowledge is a big obstacle.... even if it would "work" having the `!region` for `interpolateRect`, it doesn't mean it's actually reflective of facts. So what I'm wondering for those conditions is what is actually true... i.e. does `interpolateRect` logically require that `region` is falsey or can it be either, etc.

mcous, 2019-08-06:
If it helps at all, this is the parse "tree" of the file above:

```js
[ { type: 'set', line: 0, prop: 'nota', value: 'A' },
  { type: 'set',
    line: 0,
    prop: 'epsilon',
    value: 0.00015000000000000001 },
  { type: 'set', line: 1, prop: 'units', value: 'in' },
  { type: 'level', line: 7, level: 'polarity', value: 'D' },
  { type: 'set', line: 8, prop: 'region', value: true },
  { type: 'op', line: 9, op: 'move', coord: { x: -1, y: -1 } },
  { type: 'set', line: 10, prop: 'mode', value: 'i' },
  { type: 'op', line: 10, op: 'int', coord: { y: 1 } },
  { type: 'op', line: 11, op: 'last', coord: { x: 1 } },
  { type: 'op', line: 12, op: 'last', coord: { y: -1 } },
  { type: 'op', line: 13, op: 'last', coord: { x: -1 } },
  { type: 'set', line: 14, prop: 'region', value: false },
  { type: 'level', line: 15, level: 'polarity', value: 'C' },
  { type: 'set', line: 16, prop: 'region', value: true },
  { type: 'op', line: 17, op: 'move', coord: { x: 0.5, y: -0.5 } },
  { type: 'set', line: 18, prop: 'mode', value: 'i' },
  { type: 'op', line: 18, op: 'int', coord: { y: 0 } },
  { type: 'set', line: 19, prop: 'arc', value: 'm' },
  { type: 'set', line: 20, prop: 'mode', value: 'ccw' },
  { type: 'op',
    line: 20,
    op: 'last',
    coord: { x: 0.0054, y: 0.5, i: -0.5, j: 0 } },
  { type: 'op',
    line: 21,
    op: 'last',
    coord: { x: -0.5, y: 0, i: -0.0054, j: -0.5 } },
  { type: 'set', line: 22, prop: 'mode', value: 'i' },
  { type: 'op', line: 22, op: 'last', coord: { y: -0.5 } },
  { type: 'op', line: 23, op: 'last', coord: { x: 0.5 } },
  { type: 'set', line: 24, prop: 'region', value: false },
  { type: 'level', line: 25, level: 'polarity', value: 'D' },
  { type: 'done', line: 26 } ]
```

I think you'll also find section 2 of the [Gerber specification (PDF link)](https://www.ucamco.com/files/downloads/file/81/The_Gerber_File_Format_specification.pdf?af1eaff2d3a65450d009bc96418d5608) useful.

This function `interpolate` is reacting to the `type: 'op', op: 'int'` objects above (and the `op: 'last'` where the last operation was `int`), with the `type: 'set'` objects determining the values for `mode` and `region`. The return value for `interpolate` is a bounding box of the graphics drawn.

> Does this part have to be placed after the mode === 'i' condition?
> https://github.com/tracespace/tracespace/blob/156a00bbc350aa05f0025ff1b77090718436bc3b/packages/gerber-plotter/lib/_operate.js#L384-L388

Yes it has to be after. [`mode`](https://github.com/tracespace/tracespace/blob/master/packages/gerber-parser/API.md#set-objects) can be one of `i` (meaning linear interpolation mode), `cw` (meaning clockwise arc interpolation mode), or `ccw` (conterclockwise...).

By the time we get to L384, we know we've skipped `i`, so we must be left with one of the arc interpolation modes. According to the spec, only circular tools (there are 4 primitive tool shapes available as well as complex macro shapes) may draw arcs. Furthermore, only circular and rectangular tools may draw lines.

`tool.trace` is an array that is empty if the shape is not allowed to be used for an interpolation (or a "trace"), or contains the tools [shape parameters](https://github.com/tracespace/tracespace/blob/master/packages/gerber-parser/API.md#shapes-and-parameters) to be used to determine trace width. By checking that the trace parameters are of length 1, we are (convolutedly) checking that the tool is a circle

So, to sum up:

- `region === true`, tool doesn't matter, just call `drawLine` or `drawArc` according to `mode`
    - Check out section "4.12 Region Statement (G36/G37)" in the spec for more information about what regions are
- `region === false`, we making a trace, so these rules apply:

|   | `mode === 'i'` | `mode === 'cw'`  or `mode === 'ccw'` |
| --- | --- | --- |
| circle (`tool.trace.length === 1`) | allowed | allowed | 
| rectangle (`tool.trace.length === 2`) | allowed | not allowed |
| other (`tool.trace.length === 0` or `tool == null`) | not allowed | not allowed |

> Maybe if I had some clarity on the necessary conditions for these calls, I could refactor it in a pleasant way. I doubt this is accurate, but it's the best I could do as a starting place:

I certainly appreciate any cleanup, but I wouldn't put too much effort into this particular place in the code. It's very old and I will more than likely throw the entire `gerber-parser` module out with the v5 release... Not that I have an actual timeline for that because I work on this in my limited spare time.

> ```
>   mode === 'i'
>     && (region || tool.trace.length === 1)
>   drawLine()
> ```

This works so far. If region mode is on, we don't care at all about the tool shape, because the tool path is simply defining the contour of the region

> ```
>   mode === 'i'
>     && !region
>     && (tool.trace.length === 0 || tool.trace.length > 1)
>   interpolateRect()
> ```

This doesn't work because if `tool.trace.length` is 0, this tool is not a rectangle, so calling `interpolateRect` would be incorrect. The correct move if `tool.trace.length === 0` and `region === false` would be to warn the user about an illegal operation and move on.

The `!region` here is also redundant because it's implied by the first condition.

> ```
>   mode !== 'i'
>     && (region || tool.trace.length === 1)
>   drawArc()
> ```

This looks correct, provided there are appropriate warnings for attempting to arc a rectangular tool in non-region mode

m59peacemaker, 2019-08-06:
Wow, that is quite an informative reply! I'll try to give it another go tomorrow. Thanks!