tracespace/tracespace #84: Rendering routed panel with mousebites

ju5t, 2018-08-07:
Unfortunately I can't share the gerber files (or final renders) in public. 

It's a square panel with 4 round PCBs and 4 small square PCBs, routed around the edges and connected with mousebites. This is pretty common for us.

An example of the outline from the tracespace viewer:

[image]

We're trying to use pcb-stackup to create a top and bottom view. 

- maskWithOutline == false, blue in the screenshot becomes black, instead of the soldermask.
- maskWithOutline == true, it cuts out the PCBs completely.

The next part might be a separate issue.

The border of the PCBs has a gold finish (along with the pads, etc.) but for some reason the border  isn't covering the entire circle on the 2nd and 4th PCB in the panel. It's covering 90-360 degrees of the PCB, the first 90 degrees isn't there.

I can replicate this on the viewer. I can send the files via e-mail and I'm happy to provide more feedback if needed.

kasbah, 2018-08-07:
Hey, I am a bit confused about the image you posted. I don't see any tabs/mousebites and can't tell what's PCB and what is cut-out. 

If you send me the files privately via email or gitter I can take a look though I can't promise I'll be able to solve it of course.

ju5t, 2018-08-07:
It turns out something is off the outline file that we have. The screenshot in the initial post shows the outline in the tracespace viewer. I haven't been able to replicate this yet in gerbv or gerbview, they both show the outline as normal.

`$ gerbv -o test.png -x png contour.gko` 

results in:

[image]

I haven't received any feedback yet on which software was used to generate the outline.

kasbah, 2018-08-07:
I believe tracespace viewer tries to render the outline layer as a filled in contiguous shape so that's why it looks different.

ju5t, 2018-08-07:
I got some feedback and it comes down to 'why on earth are you using these old files'. 

I converted `contour.gko` (import/export in another program) and it now renders properly in the tracespace viewer. For some reason it didn't occur to me today that I could try this with all files too, as we're still missing 1/4th of finish in PCB two and four in the mentioned panel. I'll try that tomorrow and give those files a go.

kasbah, 2019-02-23:
I guess we can close this issue?

mcous, 2019-02-24:
@ju5t we can re-open this issue (or file new ones) if there's anything actionable here, but gonna close this one and chalk it up to bad Gerbers for now

closed 2019-02-24 by mcous