tracespace/tracespace #89: Mirror gerber to svg ?

darkworks, 2018-10-29:
hi thanks for amazing project , 

i want to know can we do mirroring in this gerber2svg package 
if yes how ? is i search for mirror but have not found anything in repo

thanks

mcous, 2018-10-29:
Hi @darkworks, mirroring is possible by adding either the `transform` SVG attribute on the main `<g>` node (I don't think we expose any hooks to do this, so it might be a little tricky) or by applying a CSS transform. Can you tell me a little bit more about the mirroring you're trying to accomplish? Also, are you using `gerber-to-svg` in JS via its API or are you using the `@tracespace/cli` tool?

darkworks, 2018-10-29:
am using old version 

[bin\g2svg\node_modules\.bin\gerber2svg.cmd --out '.$destination_file.' '.$source_path;](url)

soon will update it to the new version , 
well am trying to mirror silk bottom svg , so i think it will be good if have mirror flag in it , so that we can mirror bottom silk layers ,   so this mean for now i have to parse svg via php xml class and have to it transform to it

mcous, 2018-10-29:
I do not think we'll add a mirror option because it would be another thing to maintain. If you upgrade to the new utility:

```shell
npm install @tracespace/cli@next
```

You can flip in X using CSS with:

```shell
tracespace --noBoard --gerber.attributes.style="transform:scale(-1,1)" ...
```

(Not sure what your input/output requirements are so omitted them from the example above)

mcous, 2018-11-03:
@darkworks the example above seems to work for me, so I'm going to close this issue. Please let me know if it doesn't work for you for whatever reason and we can reopen this ticket or file a new one as necessary

closed 2018-11-03 by mcous