tracespace/tracespace #95: Is the tool expected to fully support GerberX2 file format?

justinmassiot, 2018-11-29:
`gerbv` seems to not support some instructions related to the (newer) GerberX2 format. Is this the case with tracespace?

mcous, 2018-11-29:
tracespace is primarily concerned at the moment with image-related instructions. The [API documentation of the parser](https://github.com/tracespace/tracespace/blob/next/packages/gerber-parser/API.md#transform-stream-objects) _may_ offer some guidance there.

Glancing over the latest Gerber spec, the only image related things it doesn't support are the newer `L*` commands: `LM` (mirror), `LR` (rotate), and `LS` (scale) commands.

tracespace does not support any of the metadata attributes. I would like to support some of the file attributes (e.g. `.FileFunction`) at some point, but I would probably re-write the parser before trying to extend it with attributes.

Are there any specific instructions you were wondering about?

mcous, 2018-11-29:
This tool _should_ correctly render any valid RS-274X (aka Extended Gerber) file, so I think you'll be good!

According to [Wikipedia](https://en.wikipedia.org/wiki/Gerber_format#Extended_Gerber), the difference between Gerber X1 and Gerber X2 is only the metadata attributes I mentioned above. As the metadata attributes do not affect image generation, they will be ignored.

Please do open issue(s) if you find any files that do not correctly render

closed 2018-11-29 by mcous