tracespace/tracespace #113: Add 3D view

kasbah, 2019-03-11:
As [discussed in #104](https://github.com/tracespace/tracespace/pull/104#issuecomment-467037146) using WebGL in the `view` app could improve zoom performance as well as allow for a 3D rendering of boards. 

Three.js is probably the most widely used lib for WebGL. There is also a [Three.js SVG Loader](https://threejs.org/docs/#examples/loaders/SVGLoader)
and a [React renderer for Three.js](https://github.com/drcmda/react-three-fiber).

mcous, 2019-03-12:
Looking at [the code of Three.js's example SVG loader](https://github.com/mrdoob/three.js/blob/master/examples/js/loaders/SVGLoader.js), it doesn't support the output of `gerber-to-svg` as it currently behaves (related to #80), and would require - from the looks of it - writing a completely custom Three.js-based pcb-stackup.

@henrikekblad if you don't mind, how did OpenHardware.io end up implementing their 3D view?

henrikekblad, 2019-05-08:
Sorry for a late reply (missed the github notification).

I used three.js in a somewhat creative way. Punching holes using the transparent layer. Here's a gist of the page to get you started:
https://gist.github.com/henrikekblad/292349f928fe67ff574cb975fdaa7488

kasbah, 2019-05-31:
Just leaving this svg to 3d demo here in case it's helpful in the future: https://kovacsv.github.io/JSModeler/documentation/examples/svgto3d.html

kasbah, 2020-01-20:
@ozel and I made some progress getting JSModeler to work for this. We managed to render one layer of the [DIY particle detector](https://kitspace.org/boards/github.com/ozel/diy_particle_detector/diy%20electron%20detector/) by inlining all the `<use>` tags using a [custom svgo](https://github.com/kitspace/svgo) and loading that.

https://codepen.io/kasbah/pen/QWwJyVR

[image]

jamesadevine, 2020-11-16:
Hiya, sorry to resurrect this issue, but I have a question about gerber-to-svg's SVG compatibility.

I'm not especially familiar with the SVG format as a whole, but as mentioned above Three js is not able to render resulting SVGs correctly. Would someone be able to identify the particular problem area here? Is it because of the majority of elements are defined in `<defs>`? Heavy usage of `<use>` tags? Colours not inlined on each path?

@kasbah would you be able to point me to the custom svgo plugin you used? (I know this was a long time ago :smile:)

kasbah, 2020-11-16:
The repo for my custom plugin is linked above. It's pretty much just this change: https://github.com/kitspace/svgo/commit/c31227957b340246a62053ddd045f6b976b8d29d can't remember what that was about honestly. Maybe we should try and upstream it.

jamesadevine, 2020-11-16:
@kasbah doh, thanks for pointing that out! I mistook your link for one that points to mainline svgo.

shawon1220, 2022-07-10:
@kasbah  @mcous  
I render svg to canvas and then use three.js's texture map to achieve 3D effect
[image]

shawon1220, 2022-08-03:
@mcous @kasbah Is there a good solution for tracespace to achieve 3d preview? 
This is a nice looking 3d renderer that supports outlines. It is only imperfect in layer analysis, and supports standard gerber rendering.

  demo:
  https://mayhewlabs.com/webGerber/?demo=Go_Between_Shield
  sources:
  https://github.com/KentuckySolarCar/3D-Gerber-Viewer

shawon1220, 2022-08-23:
Hello @mcous . I am using webgl to load the glb 3d model to the pcb board, but the coordinate file provided by the user, I don't know how to convert the screen coordinates where the upper bit number of top.svg is located, please help, thank you.

The format of the coordinate file provided by the user is as follows:

Designator  	Layer	      Center-X(mm)	    Center-Y(mm)	Rotation
C1                   TopLayer	       97.5799	            140.644	        0
C2                   TopLayer	       97.5041	             147.1441             180
...