tracespace/tracespace #178: Publish `<script>` bundles to npm (for usage with unpkg)

mcous, 2019-03-23:
As a potential tracespace user, I would like to be able to try the libraries out (or use them) via a `<script>` tag so that I don't need a bundler in my workflow if I don't want to.

We should publish bundles for the following libraries:

- gerber-parser
- gerber-plotter
- gerber-to-svg
- whats-that-gerber
- pcb-stackup-core
- pcb-stackup

Thanks to @galacticstudios for the request!

galacticstudios, 2019-03-23:
Publishing <script> bundles is one approach. Another, perhaps better one, would be to write up a dummies guide to bundling up the libraries. It's like the old saying about teaching a man to fish.

mcous, 2019-03-23:
More concrete browser usage examples in the docs is a good idea. Will definitely make it a priority

PR #180 (2019-03-23): build: Build/publish browser bundles for `<script>` usage
This PR adds bundle builds to all external-facing libraries so that they can be used with `<script>` tags. Closes #178
commit ef475456 (2019-03-23): build: Build/publish browser bundles for `<script>` usage (#180)

Closes #178
closed 2019-03-23 by mcous