tracespace/tracespace #215: NPM Issue

creamdory, 2019-04-09:
OS : Ubuntu 18.04 LTS

When I am trying to setup tracespace in our server, I am getting this during `nom run start`

```
gerber-to-svg: events.js:183
gerber-to-svg:       throw er; // Unhandled 'error' event
gerber-to-svg:       ^
gerber-to-svg: Error: listen EADDRINUSE :::8002
gerber-to-svg:     at Object._errnoException (util.js:1022:11)
gerber-to-svg:     at _exceptionWithHostPort (util.js:1044:20)
gerber-to-svg:     at Server.setupListenHandle [as _listen2] (net.js:1367:14)
gerber-to-svg:     at listenInCluster (net.js:1408:12)
gerber-to-svg:     at Server.listen (net.js:1492:7)
gerber-to-svg:     at Function.listen (/root/tracespace/node_modules/express/lib/application.js:618:24)
gerber-to-svg:     at Object.<anonymous> (/root/tracespace/packages/gerber-to-svg/integration/server.js:10:47)
gerber-to-svg:     at Module._compile (module.js:652:30)
gerber-to-svg:     at Object.Module._extensions..js (module.js:663:10)
gerber-to-svg:     at Module.load (module.js:565:32)
gerber-to-svg: npm ERR! Linux 4.15.0-47-generic
gerber-to-svg: npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "run" "start"
gerber-to-svg: npm ERR! node v8.10.0
gerber-to-svg: npm ERR! npm  v3.5.2
gerber-to-svg: npm ERR! code ELIFECYCLE
gerber-to-svg: npm ERR! gerber-to-svg@4.0.2 start: `node integration/server`
gerber-to-svg: npm ERR! Exit status 1
gerber-to-svg: npm ERR!
gerber-to-svg: npm ERR! Failed at the gerber-to-svg@4.0.2 start script 'node integration/server'.
gerber-to-svg: npm ERR! Make sure you have the latest version of node.js and npm installed.
gerber-to-svg: npm ERR! If you do, this is most likely a problem with the gerber-to-svg package,
gerber-to-svg: npm ERR! not with npm itself.
gerber-to-svg: npm ERR! Tell the author that this fails on your system:
gerber-to-svg: npm ERR!     node integration/server
gerber-to-svg: npm ERR! You can get information on how to open an issue for this project with:
gerber-to-svg: npm ERR!     npm bugs gerber-to-svg
gerber-to-svg: npm ERR! Or if that isn't available, you can get their info via:
gerber-to-svg: npm ERR!     npm owner ls gerber-to-svg
gerber-to-svg: npm ERR! There is likely additional logging output above.
gerber-to-svg: npm ERR! Please include the following file with any support request:
gerber-to-svg: npm ERR!     /root/tracespace/packages/gerber-to-svg/npm-debug.log
lerna ERR! npm run start exited 1 in 'gerber-to-svg'
lerna ERR! npm run start stdout:

> gerber-to-svg@4.0.2 start /root/tracespace/packages/gerber-to-svg
> node integration/server

lerna ERR! npm run start stderr:
events.js:183
      throw er; // Unhandled 'error' event
      ^

Error: listen EADDRINUSE :::8002
    at Object._errnoException (util.js:1022:11)
    at _exceptionWithHostPort (util.js:1044:20)
    at Server.setupListenHandle [as _listen2] (net.js:1367:14)
    at listenInCluster (net.js:1408:12)
    at Server.listen (net.js:1492:7)
    at Function.listen (/root/tracespace/node_modules/express/lib/application.js:618:24)
    at Object.<anonymous> (/root/tracespace/packages/gerber-to-svg/integration/server.js:10:47)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)

npm ERR! Linux 4.15.0-47-generic
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "run" "start"
npm ERR! node v8.10.0
npm ERR! npm  v3.5.2
npm ERR! code ELIFECYCLE
npm ERR! gerber-to-svg@4.0.2 start: `node integration/server`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the gerber-to-svg@4.0.2 start script 'node integration/server'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the gerber-to-svg package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node integration/server
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs gerber-to-svg
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls gerber-to-svg
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /root/tracespace/packages/gerber-to-svg/npm-debug.log

lerna ERR! npm run start exited 1 in 'gerber-to-svg'
lerna WARN complete Waiting for 3 child processes to exit. CTRL-C to exit immediately.

npm ERR! Linux 4.15.0-47-generic
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "run" "start"
npm ERR! node v8.10.0
npm ERR! npm  v3.5.2
npm ERR! code ELIFECYCLE
npm ERR! tracespace@ start: `lerna run start --parallel`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the tracespace@ start script 'lerna run start --parallel'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the tracespace package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     lerna run start --parallel
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs tracespace
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls tracespace
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /root/tracespace/npm-debug.log
```

mcous, 2019-04-09:
Thanks for the report @creamdory! Looks like a race condition in automatic port selection for the dev servers. I'll see what can be done about it. To work around, I would recommend launching the dev servers individually:

```shell
# run only the @tracespace/view development server
npm start -- --scope @tracespace/view
```

Details here: <https://github.com/tracespace/tracespace#development-servers>

creamdory, 2019-04-09:
Thanks @mcous ! I will take a look and try :)

mcous, 2019-04-10:
@creamdory I re-read your error log, and I realized that I had the wrong script on my mind when I was talking about automatic port selection. For `npm start`, there is no automatic port selection; the development server ports are hardcoded as:

- gerber-to-svg: `8002`
- pcb-stackup-core: `8001`
- pcb-stackup: `8000`
- @tracespace/view: `8080`

If you receive an `EADDRINUSE` error, it means something else on your machine is running on that port. Did you perhaps run the dev servers in another terminal or in a background process? Otherwise, you can use `lsof` to find which process is running on a given port: e.g. `lsof -i :8002`.

I'm going to close this issue because this does not look like a problem with the dev server script setup, but rather an actual port conflict on your machine. I'll definitely re-open the ticket if I'm mistaken though!

closed 2019-04-10 by mcous