tracespace/tracespace #220: Unable to process .zip files using Windows OS

jeremythegreat01, 2019-04-10:
Dear Tracespace View team,

We are unable to generate a proper output whenever a `.zip` file is used - could you please advise?
[image]

###### Notes
1. Using Windows OS with Chrome and Firefox browsers = same error code is encountered
 - We tried to use other .zip files = same error code is encountered
 - Unzipping the files and adding them separately works
2. Using MacOS = it works (no problems, so far)

###### Steps to re-create issue
1. Download test `.zip` file from `https://tracespace.io/view/arduino-uno.zip`
2. Click on the "Add file" button
3. Upload the `arduino-uno.zip` file
4. The `react-dom.production.min.js:102 Error: <use> attribute x: Expected length, "NaN".` error is encountered (See `Console Error Log` section for full reference)

###### Console Error Log
```
react-dom.production.min.js:102 Error: <use> attribute x: Expected length, "NaN".
nr	@	react-dom.production.min.js:102
Ha	@	react-dom.production.min.js:227
Qa	@	react-dom.production.min.js:232
Ka	@	react-dom.production.min.js:233
zo	@	react-dom.production.min.js:249
Po	@	react-dom.production.min.js:248
wo	@	react-dom.production.min.js:245
Ga	@	react-dom.production.min.js:243
oi	@	react-dom.production.min.js:157
(anonymous)	@	store.ts:20
l.current.onmessage	@	store.ts:56
```

mcous, 2019-04-10:
Thanks for the report! Looks like on windows, the ZIP file is coming in with MIME type `application/x-zip-compressed`, while the app is looking for `application/zip` to identify ZIP files. This is an easy fix, so I should have it up shortly

closed 2019-04-10 by mcous

dottenbr, 2019-04-10:
@mcous Thank you for this fast fix, when will this be live on tracespace.io?
@jeremythegreat01 please confirm this fixes the problem.

mcous, 2019-04-10:
`v4.0.3` is running through CI now and should be live in a few minutes! Thanks again for the report

dottenbr, 2019-04-10:
We can confirm this actually fixed the problem 👍

jeremythegreat01, 2019-04-10:
Hello @dottenbr, Hello @mcous, 

I confirm that the `.zip` file issue is no longer being encountered while using the Windows OS.

Note: Tested with Chrome and Firefox browsers as well.

Thank you for the quick fix!

PR #222 (2019-04-10): fix(view): Make ZIP logic more lenient by checking filename
On Windows, ZIP files coming in via FileReader have a different MIME type than on MacOS, causing the file to be sent straight to the render logic rather than through the unzipper.

Clearly MIME type should not be relied on, so this PR make the "is this a ZIP file" logic more lenient by:

- Simply looking at the extension first to see if it ends in `.zip`
- If it doesn't or there's no filename, checking the MIME type against a few possible values

Fixes #220
commit a37a02d9 (2019-04-10): fix(view): Make ZIP logic more lenient by checking filename (#222)

Fixes #220