tracespace/tracespace #243: how can get more information of PCB ? eg. width,height

aaronbo2019, 2019-04-30:
how can get more information of PCB ? eg. width,height
I need show the width and height of PCB in browser. Which file need modify,how modify?

mcous, 2019-04-30:
Hi @aaronbo2019, are you using the pcb-stackup library to convert your files? If so, the width and height are available here:

```js
pcbStackup(layers).then(stackup => {
  console.log(stackup.top.width)  // number
  console.log(stackup.top.height)  // number
  console.log(stackup.top.units)  // 'in' | 'mm'
  // also available on the `bottom` object with the same values
  console.log(stackup.bottom.width)
  console.log(stackup.bottom.height)
  console.log(stackup.bottom.units)
})
```

Docs for the stackup object: https://github.com/tracespace/tracespace/blob/master/packages/pcb-stackup-core/README.md#usage

aaronbo2019, 2019-04-30:
thank you for your reply！

mcous, 2019-04-30:
No problem. If you have any more questions, please feel free to reach out via [Gitter](https://gitter.im/tracespace/Lobby)

closed 2019-04-30 by mcous

Gianarci, 2024-02-16:
I've the same question but without using pcb-stackup library that is discontinued. How can I get size and unit?