tracespace/tracespace #247: How define the "layers"?

aaronbo2019, 2019-05-03:
How define the "layers"? My back-end server is based on PHP.
In html front，cannot  use  **fs.createReadStream**(node ),really annoyed.

use a script tag:
```html
<script src="https://unpkg.com/pcb-stackup-core@^4.0.0/dist/pcb-stackup-core.min.js"></script>
<script>
  const layers = GERBER_FILENAMES.map(filename => ({
  filename,
  gerber: fs.createReadStream(path.join(GERBERS_DIR, filename)),
}))
  var stackup = pcbStackupCore(layers)
</script>

mcous, 2019-05-03:
Hi @aaronbo2019, I'm sorry you're annoyed. We don't currently have any examples up for browser-side usage, so here's a few tips:

- I _highly_ recommend you **use pcb-stackup**, not pcb-stackup-core
    - pcb-stackup-core is for advanced usage and requires you to do all the orchestration work that pcb-stackup just handles
- The shape of a `Layer` is detailed in the [pcb-stackup docs](https://github.com/tracespace/tracespace/blob/master/packages/pcb-stackup/API.md#layer)
- The simplest possible layer would be an object with a `filename` property with the name of the file and a `gerber` property with the **full contents** of the file:
    ```js
    const myLayer = {filename: 'arduino-uno.cmp', gerber: 'G75*\n%MOIN*%\n%OFA0B0*%\n...'}
    ```

**How you get the file contents to the browser is up to you** and completely outside the scope of the tracespace libraries. If, for example, you wanted to use [`FileReader`](https://developer.mozilla.org/en-US/docs/Web/API/FileReader) (like we do in @tracespace/view), something like this could work:

```html
<!DOCTYPE html>
<html>
  <head>
    <meta charset="UTF-8" />
    <title>pcb-stackup</title>
  </head>
  <body>
    <input class="input" type="file" multiple />
    <div class="output"></div>
    <script src="https://unpkg.com/pcb-stackup@^4.0.0/dist/pcb-stackup.min.js"></script>
    <script>
      const $in = document.querySelector('.input')
      const $out = document.querySelector('.output')

      $in.addEventListener('change', event => {
        const files = Array.from(event.target.files)
        const layersCount = files.length
        const layers = []

        event.target.value = ''

        files.forEach(file => {
          const reader = new FileReader()

          reader.addEventListener('load', loadEvent => {
            layers.push({
              filename: file.name,
              gerber: loadEvent.target.result,
            })

            if (layers.length >= layersCount) {
              pcbStackup(layers).then(
                stack => ($out.innerHTML = stack.top.svg + stack.bottom.svg)
              )
            }
          })

          reader.readAsText(file)
        })
      })
    </script>
  </body>
</html>
```

(Please note that I've written the above example in ES6 for ease. It won't work in every single browser)

closed 2019-05-03 by mcous

aaronbo2019, 2019-05-04:
@mcous thanks! I have completed it using your example.
[image]