tracespace/tracespace #251: null null in output svg name

darkworks, 2019-05-10:
hi when i try to convert single gerber using below command i get null null in name , i wonder can we supply output name to command or i think will be good if get same file name as input file name because if we get nill null in output name it will be hard to trace output file through programming 

`tracespace --gerber.attributes.color=#0366d6 --gerber.attributes.style="transform:scale(-1,1)" --force --noBoard --out c:\renderings c:\silk_bot.art`

mcous, 2019-05-10:
Thanks for filing an issue on this, it is a kind of annoying thig the CLI does when whats-that-gerber can't figure out an input filename. In this case, the first `null` means unknown board side, and the second one means unknown layer type. If `silk_box.art` is something that should be recognized, would you mind also filing an issue to get it recognized in whats-that gerber? See #246 for a good example of that kind of issue.

If you're interested in opening a PR for the CLI to get rid of the `null`s, the output filename is set here:

https://github.com/tracespace/tracespace/blob/e1cd3af69766ae2f08491cb2309a62013ff1f815/packages/cli/lib/cli.js#L117-L120

I'm thinking the CLI should skip appending `.side` and/or `.type` if they evaluate to `null`.

I also think some customization of the output filename would be really cool, but I'm unsure out to best accomplish it. Maybe some sort of output filename template option? Might need to think about filename customization specifically a little bit longer

darkworks, 2019-05-13:
ok i have fixed currently by removing null from file name string , but in future will be good if we have output file name parameter

i will send PR request for it

commit 09f1c4e4 (2019-06-03): fix(cli): Do not append "null" to output filenames (#253)

Closes #251
closed 2019-06-03 by mcous