tracespace/tracespace #278: Unhandled promise rejection  at Firefox v67.0.4

Summving, 2019-06-29:
as below image
[image]

mcous, 2019-06-29:
Thanks for the report @Summving! Are you able to get any of the following?

- Any problems with the site's behavior when you saw this error
- Which operating system you're on
- A longer stack trace if you have it
- Any steps to reproduce

Summving, 2019-07-10:
> 
> 
> Thanks for the report @Summving! Are you able to get any of the following?
> 
>     * Any problems with the site's behavior when you saw this error
> 
>     * Which operating system you're on
> 
>     * A longer stack trace if you have it
> 
>     * Any steps to reproduce

Hi, There is no steps, just open this link, and click the  Privacy Policy Done botton, it show these errors, and the privacy dialog can not be closed. Have tried at incognito mode, still same.
Windows 10, 64bit
after I uninstalled Firefox 67 and installed Firefox 68, this issue was gone.
Maybe it something wrong at previous Firefox.
Thanks

closed 2019-07-10 by Summving

mcous, 2019-07-10:
Thanks for the extra information! Upon investigation:

- The unhandled promise rejection was coming from Dexie.js (dfahlander/Dexie.js#831)
- This is due to a bug in Firefox that breaks IndexDB **only when the dev console is open** (https://bugzilla.mozilla.org/show_bug.cgi?id=1545400)

So you're correct, it was a Firefox 67 issue, but only with the dev console open. If you close the dev console, Firefox 67 works fine

Summving, 2019-07-10:
emmm...
I found it didnt work , and then open the dev console for checking.....