tracespace/tracespace #298: Mask swell & paste shrink

stg, 2019-09-18:
This is a fantastic viewer!

It would be slightly more fantastic if there was an option to swell the soldermask and shrink the paste mask by configurable values.

I think this could be implemented by simply adjusting gerber apertures before plotting.
Rectangular and circular apertures would be trivial, but I suppose polygonal apertures would be more tricky.

mcous, 2019-10-07:
Thanks for the feature request! I think this will be easier whenever [v5](https://github.com/tracespace/tracespace/tree/v5) is ready, because the parse, plot, and render trees will all be actual, navigable trees (as opposed to streams in the current implementation).

This would be easier if we could use a simple SVG `scale` transform at some percentage value (e.g. 1%). Some cursory Googling seems to lead me to mask swells spec'd in mils, so I wonder if a percentage scale would be satisfactory or not

stg, 2019-10-07:
My take is that a percentage would not suffice for accurate rendering, as this would result in a significantly different swell/shrink for a large pad vs a small pad and also cause unevenness for oval or rectangular pads.

I am not well versed in gerber, but as I had understood it the format revolves mostly around plotting apertures along lines or arcs? I believe the correct place to implement this is by changing the dimensions of the aperture before plotting. For example a circular aperture of 2mm would become a circular 2.2mm aperture with a swell parameter of 0.1mm. A rectangular 2x1mm aperture would become a 2.2x1.2mm aperture. This swell/shrink could be a parameter for rendering each specific layer, defaulting to zero for all layers except mask and paste.

Factories typically adjust shrink and swell as well as do other modifications to these layers for production - but basic shrink and swell would go a long way towards producing accurate results. The next most common difference between gerber and physical board is in my experience the removal of thin "soldermask bridges" (google) for fine pitched SMD components.

I see you have already taken care of silkscreen removal over pads by proper layer ordering, which is very nice indeed - although you may want to consider this an option, as it is not always done, nor is it always wanted (although likely a percentile issue at most).

mcous, 2019-10-07:
Thanks for the detailed reply! It seems like:

- Percentage-based shrink/swell is not an acceptable solution
- Circle, rectangle, and obround apertures should be fairly easy to swell
- Regular polygon apertures shouldn't be too bad, because each point can be moved away from the center, but will be more involved than the "easy" shapes
- Aperture macros are probably too much trouble for an initial implementation

Shrink and swell can be added as an option to the v5 plotter, which can add a `data` parameter to the relevant `shape` nodes in the plot tree with the augmented dimensions