tracespace/tracespace #302: Board outline is incorrect

attie, 2019-10-08:
I've got a couple of boards that have a somewhat complex outline, and unfortunately they don't render well...

- [bad_outline.gm2.txt](https://github.com/tracespace/tracespace/files/3704586/bad_outline.gm2.txt) (remove `.txt` extension)

I suspect that this is an issue with drawing filled SVG Paths, and the fact that this outline is actually rendered as a number of `<path>` elements, but I'm not fully sure. I imagine that rendering the board outline as a single path would resolve this issue.

Extract from `top.svg`: (lines are truncated)

```svg
<path d="M 100380.3 -28920.1 100379.4 -28920.1 100378.6 -28920.9 100376.1 -28921.8 100371.9 -28922.6 10
<path d="M 99766.3 -28587.7 99583 -28587.7 99583 -28816.8 99755.4 -28816.8 99761.3 -28816 99767.9 -2881
<path d="M 95174.9 -28582.7 94960 -28582.7 94960 -28795.1 95155.8 -28795.1 95161.6 -28794.3 95174.9 -28
<path d="M 98730.7 -28743.5 98724.9 -28743.5 98720.7 -28744.3 98715.7 -28745.1 98709.9 -28746 98696.5 -
<path d="M 98233.3 -28743.5 98227.5 -28743.5 98223.3 -28744.3 98218.3 -28745.1 98212.5 -28746 98199.2 -
<path d="M 96943.6 -28743.5 96938.6 -28743.5 96934.5 -28744.3 96930.3 -28744.3 96924.5 -28745.1 96912 -
<path d="M 95698.1 -28743.5 95692.3 -28743.5 95688.1 -28744.3 95683.1 -28745.1 95677.3 -28746 95664 -28
<path d="M 96382.1 -29152.5 96312.1 -29152.5 96312.1 -29084.2 96311.3 -29085.1 96309.6 -29087.6 96306.3
<path d="M 100908.5 -28742.7 100909.3 -28741.8 100911 -28740.2 100914.3 -28736.8 100918.5 -28732.7 1009
```

[image]

[image]

mcous, 2019-10-09:
Thanks for the report! I gave this file a look over, and here are some initial thoughts:

- It looks like the way Altium encodes text into Gerber breaks the SVG outline clipping tracespace uses. Until we can address that somehow, I recommend that you **don't put anything other than the outline itself in your outline layer**
    - More specifically, this Gerber file appears to draw text, and then clear away holes in the text
    - It's this clearing layer in the outline Gerber that breaks tracespace's outline strategy completely
- Our arc handling in v4 is a little wonky (see #82), and this outline _may_ have some small gaps in it (either inherently, or as a result of wonky arc plotting)

In tracespace, I can get this outline to render properly if I do the following:

- Remove the text from the Gerber because it breaks outline rendering
- Set the "Gap fill limit" in the board settings to 0.14 mm

attie, 2019-10-09:
> It looks like the way Altium encodes text into Gerber breaks the SVG outline clipping tracespace uses. Until we can address that somehow, I recommend that you don't put anything other than the outline itself in your outline layer

Oh, I didn't even think about that! Thanks very much for looking in to the issue - I'll keep that in mind.

I'll leave the ticket open (I feel it's something worth addressing - lots / all of my boards have text outside to identify the layer), but if you disagree, please feel free to close it.

mcous, 2019-10-09:
Did some more investigating, and this is a very interesting pair of issues! I really appreciate you taking the time to write them up and bearing

There is _probably_ some interaction with #302 going on here, but the most pressing issue I can find here is that **in Firefox** (my browser of choice), clear masks in the outline `<clipPath>` break the outline entirely, whereas in Chrome the `clipPath` still works. I agree that this ticket should stay open for that reason.

In terms of the other stuff:

- #82 is _probably_ in play here, but luckily arc plotting has already been completely revamps in the [v5 development branch](https://github.com/tracespace/tracespace/tree/v5), so hopefully we see some improvements there
- Gaps in the outline (excepting ones caused by wonky arc plotting) are pretty common, and I'm certainly trying to make improvements in that area in v5, I don't have anything concrete yet. In the mean time, I definitely recommend playing with the "gap fill limit" option whenever an outline doesn't render or doesn't fill out properly

attie, 2019-10-10:
Hey, no problem - thanks for such a great response!

Just to clarify, I've tried removing the text / layer label from the gerber, and I'm seeing the same issue.
I edited the gerber rather than the source document and re-exporting.

- [bad_outline_no_text.gm2.txt](https://github.com/tracespace/tracespace/files/3714529/bad_outline_no_text.gm2.txt)

## Chrome 77.0.3865.90

- `bad_outline.gm2`
  - "_layers_" view: fine
  - "_top_" view: broken (as [above](https://github.com/tracespace/tracespace/issues/302#issue-504294199))
- `bad_outline_no_text.gm2`
  - "_layers_" view: fine
  - "_top_" view: broken (as [above](https://github.com/tracespace/tracespace/issues/302#issue-504294199))

## Firefox 69.0.3

- `bad_outline.gm2`
  - "_layers_" view: fine
  - "_top_" view: broken (_nothing_ is rendered)
- `bad_outline_no_text.gm2`
  - "_layers_" view: fine
  - "_top_" view: broken (as [above](https://github.com/tracespace/tracespace/issues/302#issue-504294199))

mcous, 2019-10-10:
@attie, sorry I should've been more clear, because there's a few things going on:

- Nothing at all being rendered in Firefox: **bug**
- Outline fill being funny as in original post: **not a bug**

The fill is funny because the lines and arcs, as described in the Gerber file (and _possibly_ as drawn in Altium), have tiny gaps. For example, here's an extreme zoom with the tool size modified:

[image]

tracespace can fill gaps to a point, but at the end of the day (and for the v4 release line) I don't think you can expect it to work in all cases of "bad" data.

In this case, I would recommend:

- Play with the "gap fill limit" outline in board settings (in this case, a value of 0.14mm works for this board)
- See if increasing the resolution of the Gerber output helps (the Gerber spec itself recommends at least 2.5, and these files are 4.4)
- See if you can re-draw the lines to meet exactly in CAD (if it wasn't an output resolution problem)

In the v5 release line, I'd like to play with teaching tracespace how to recognize that an outline it has isn't likely to fill properly, and enable/disable the board outline render accordingly. I have no idea how that's going to happen, though, and it will likely involve me writing at least one or two new issues. I also have no idea when v5 will be ready

attie, 2019-10-12:
Hi @mcous, I understand now - thanks very much for the additional information... I'll review my outline placement!