tracespace/tracespace #303: Silkscreen has some filled voids

attie, 2019-10-08:
I don't _think_ this is the same as #302, but it may be... (it has different characteristics, but I'm leaning towards incorrect SVG Paths for both).

It seems that any / all silkscreen has filled voids that are rendered incorrectly in the "_top_" and "_bottom_" views... However the "_layers_" view and exported SVGs are fine.

[image]

[image]

[board.zip](https://github.com/tracespace/tracespace/files/3704640/board.zip)

mcous, 2019-10-09:
Thanks for the report! This is an interesting one. After investigation:

- The Gerber files in this issue and #302 use a combination of draws and clears (at the Gerber level) via `%LPD%*` and `%LPC%*` 
- tracespace re-creates these draw and clear layers using SVG masks
- As an optimation, tracespace view attempts to re-use the layer renders in the board renders with SVG `<use>` elements

These boards render correctly with the [tracespace CLI](https://github.com/tracespace/tracespace/tree/master/packages/cli), which _does not_ attempt to re-use anything. So it looks the online viewer may need to ditch the layer re-use

(The SVG paths, if you're curious, are not related. Gerber, in essence, is a vector image format. Most of what this library does is convert Gerber constructs directly into SVG constructs. tracespace takes Gerber "traces" and turns them into SVG `<path>`s, where (generally) there will be one `<path>` element for all the traces the Gerber file draws with a given tool before it changes the tool or changes the region mode)

attie, 2019-10-09:
Thanks for the update - I hope you don't mind if I leave this in your capable hands to resolve... please let me know if you'd like any additional files / testing.

closed 2019-10-10 by mcous

mcous, 2019-10-10:
Fix for this issue was released in v4.2.0!

PR #305 (2019-10-10): fix(view): Ensure layer defs are accessible to <mask>s in board renders
Previously, when one of the board renders was selected, the layers render was set to `display: none`. This made any `<mask>`s referenced in the board renders fail to load, because the `mask="url(#some_mask_id)"` attribute would no longer resolve.

This PR switches the hiding mechanism for the layers view from `display: none` to clipping.

Fixes #303
commit 0d0ee60e (2019-10-10): fix(view): Ensure layer defs are accessible to <mask>s in board renders (#305)

Fixes #303