tracespace/tracespace #311: Incorrect edge cut renders

harshitgoel96, 2019-10-17:
Hi,

I first want to thank you for providing such a nice tool to verify Gerber files.

I have notices an issue in the render, where the pcb edges are incorrect. I can see the outline absolutely fine in JLCPCB's gerber viewer. But the top edge is not properly showing for my pcb.

Image from live viewer
[image]

Image for JLCPCB viewer
[image]

mcous, 2019-10-17:
Hi @harshitgoel96, thanks for the report. Have you tried adjusting the "gap fill limit" option as suggested in the [troubleshooting guide](https://github.com/tracespace/tracespace/blob/master/apps/view/HELP.md#no-image-or-incorrect-board-shape)?

I might be able to give you more information if you were able to send me the outline file, but from looking at your screenshots, it looks like you've got some arc segments in the outline. Typically when I see incorrect board outlines involving arc segments, it's because the endpoints of the arcs don't properly meet with the endpoints of their neighboring segments, resulting in gaps in the outline that confuse the SVG fill

mcous, 2019-10-27:
Closing due to lack of response and my assumption that this is a gap-in-the-outline issue and not an unknown bug in the plotter. Will definitely re-open if I hear back and it turns out my assumption is wrong

closed 2019-10-27 by mcous

harshitgoel96, 2019-10-29:
Hi. I did use arch segment. Haven't got around to fix it, as in middle of system upgrade. Will update once I fix things up.