tracespace/tracespace #314: Is there a way to use gerber-parser to generate Gerber files from SVG files?

ceremcem, 2019-11-27:
I'm implementing Gerber export option to [aeCAD](https://github.com/aktos-io/aecad). Is there a way to re-use gerber-parser in the reverse direction to generate Gerber files from SVG (or, from some internal format, for sure)?

mcous, 2019-12-23:
Hey @ceremcem, sorry about the delay in response! There's nothing like this at the moment, as the current implementation of `gerber-parser` and `gerber-plotter` don't lend themselves well to reversing the process.

In the v5 WIP branch, I'm developing some new packages, [`@tracespace/parser`](https://github.com/tracespace/tracespace/tree/v5/packages/parser) and [`@tracespace/plotter`](https://github.com/tracespace/tracespace/tree/v5/packages/plotter) that actual define ASTs for the [Gerber format](https://github.com/tracespace/tracespace/blob/v5/packages/parser/src/tree/nodes.ts) and [an internal "image" format]https://github.com/tracespace/tracespace/blob/v5/packages/plotter/src/plot-tree/nodes.ts), respectively

It would take some work to reverse from an image AST to a Gerber, but it would definitely be possible. The main barrier is the fact that those formats are incomplete. I'm always looking for help, so let me know if any of that sounds interesting

ceremcem, 2020-01-14:
Thanks for the reply. I'd many distractions, so apologies for my late response. After digging the original documentation, it's absolutely possible to create a JS graph library like API on top of GERBER format. I'm absolutely eager to create such a thin library. Maybe a GERBER exporter to PaperJS. However, I'll currently stick to a quick-and-dirty implementation way to get the job done. In the meanwhile, we may discuss any possibility.

closed 2020-01-14 by ceremcem