tracespace/tracespace #321: My Gerber output is not rendered correctly

ceremcem, 2020-01-30:
Hi, 

I'm producing Gerber files on my own. Here is my actual PCB drawing: 

[image]

Here is the Gerber output: 
[gerber-test-gerber.zip](https://github.com/tracespace/tracespace/files/4135689/gerber-test-gerber.zip)

Here is the GerbView render: 

[image]

Here is tracespace.io render: 

[image]

How can I debug the case?

mcous, 2020-01-30:
Hi @ceremcem! Have you followed the [troubleshooting guide](https://github.com/tracespace/tracespace/blob/master/apps/view/HELP.md)? If not, I recommend doing so. If you have already, what was the result of changing the recommended options?

mcous, 2020-01-30:
The tracespace view application is a little less lenient that other gerber viewers, and bad Gerbers that may render in another application might not render here. Upon a closer inspection, it looks like your Gerber output isn't quite correct. The format specification in your file is 2.5 with leading zero suppression:

```
%FSLAX25Y25*%
```

However, in your coordinates, your X coordinates have 7 digits (good; 2+5 == 7) and your Y coordinates have 8 digits (bad; 2+5 != 7):

```
X4592896Y17580234D02*
^1234567^12345678|
```

It looks like your Gerbers are actually output at 3.5, not 2.5. Setting the "override gerber integer/decimal coordinate format" to 3.5 fixes the layer render.

As for the board layer, it looks like your Gerber output is rendering a lot of redundant layer polarity (`%LPD*%`), tool change (`D10*`), and interpolation mode (`G01*`) commands:

```
%LPD*%
D10*
X7652605Y14117353D02*
G01*
X2645Y14117353D01*

%LPD*%
D10*
X2645Y14117353D02*
G01*
X2645Y19997353D01*
```

Every time tracespace sees a layer change command, it creates a new layer, even if the polarity didn't change. This breaks the outline search functionality, which is looking for a path loops on a single drawing layer. Removing the redundant layer polarity callouts fixes the board image render for me. Removing the redundant tool changes and interpolation mode commands doesn't affect the image:

```
G04 project-name*
%FSLAX25Y25*%
%MOMM*%
%ADD10C,0.2*%
%LPD*%
D10*
X7652605Y19997353D02*
G01*
X7652605Y14117353D01*
X7652605Y14117353D02*
X2645Y14117353D01*
X2645Y14117353D02*
X2645Y19997353D01*
X2645Y19997353D02*
X7652605Y19997353D01*
M02*
```

I'm going to close this issue as the issues you are seeing are the result of bad Gerber data. Let me know if I can help with anything else!

closed 2020-01-30 by mcous

mcous, 2020-01-31:
Oops, I'm sorry! I realize now I forgot to answer your original question:

> How can I debug the case?

When I was debugging this, I followed these steps:

1. Switched to "layers" view in tracespace. Here, I noticed:
    - The shapes of pads and trace thicknesses looked correct
    - The placement of pads and traces looked good in X but compressed by about an order of magnitude in Y
2. I opened the Gerber files themselves and checked that the declared format spec (`%FS...`) matched what tracespace parsed out
    - They matched, so I moved on to
3. I examined the coordinate values themselves (`X..Y...`) and noticed the mismatch in X and Y
    - I messed around with modifying the coordinates themselves and overriding the format spec until I got the correct layer render
4. With this solved, I switched to the board view to find the outline detection still not working
5. I switched back to layer view and examined the SVG output of the outline layer
    - For a successful outline, I would expect a single `<path>` node
    - Instead I saw 4 `<path>`s, which led me to the layer polarity commands (`%LPD*%`)
    - This would've been hard to debug unless one was pretty intimately familiar with how the `gerber-plotter` package handles layer polarity commands; specifically, it ends any in-progress traces anytime it sees one
6. I removed the redundant layer polarity commands to find that the outline layer (and therefore the board) then rendered properly

ceremcem, 2020-01-31:
You helped me a lot! 

1. Regarding the `%FSLAX25Y25*%` case:
It was my mistake not to count the total integer part digits of resulting coordinates. `3.` will fit any board layout anyway. Thanks for this finding. 

2. Regarding to the excessive tool change: 

I preferred "send "export to Gerber" message to every primitive component (pad, edge, trace), let them produce their own files and then concatenate them by reducing those hundreds of Gerber files" approach in my PCB design software. However, in order not to fall into the "premature optimization trap", I skipped stripping down the redundant tool selection (and other) command entries. Apparently it was mature.