tracespace/tracespace #326: Need help to run the project

pinal3257, 2020-04-08:
Hi team,

Can you please help me to run the project?
Also where i need to set the gerber files path?

Thanks
Pinal

pinal3257, 2020-04-10:
Hi Team,

Can you please help me to run the project?
I am getting the error: "ERROR in   Error: Child compilation failed:"

Thanks
Pinal

mcous, 2020-04-10:
Hi @pinal3257! If you're looking for help setting something up with tracespace, I would recommend posting your question in the Gitter chat room: https://gitter.im/tracespace/Lobby

From your post, I'm not sure what you're trying to do, but I would recommend the following reading, as well:

- If you're trying to render Gerber files from your terminal, check out the `@tracespace/cli` README: https://github.com/tracespace/tracespace/tree/master/packages/cli
- If you're trying to render Gerber files from your web browser, just go to <https://tracespace.io/view>
- If you're trying to integrate tracespace into an existing Node.js or front-end project, check out the `pcb-stackup` README: https://github.com/tracespace/tracespace/tree/master/packages/pcb-stackup
- If you're trying to set up a local development environment to open a PR and contribute code, please read the "Contributing" section of the main README: https://github.com/tracespace/tracespace#contributing

closed 2020-04-10 by mcous

pinal3257, 2020-04-11:
Hi Team,

I want to setup my development environment and want to convert some gerber files to svg from my local development server.

Thanks
Pinal

mcous, 2020-04-13:
@pinal3257 I'm still a little confused about what you're trying to set up. If you want to simply convert Gerber files to SVG, then you do not need to set up a tracespace development environment. For conversion as part of another project, you should `npm install pcb-stackup`.

You could also explore the tracespace CLI with `npm install -g @tracespace/cli` (which is a small CLI wrapper around the `pcb-stackup` module.

If you are trying to make changes to tracespace and contribute code, I think it'll be fastest if you chat in our [Gitter chatroom](https://gitter.im/tracespace/Lobby) with details about your setup and which step(s) in the [Contributing Guide](https://github.com/tracespace/tracespace#contributing) are not working for you

pinal3257, 2020-04-13:
Hi Team

Actually i want to setup development environment.
I want to run the project in my local machine.
I install all packeges but still i am getting error when i execute "yarn start" command.
The error is : "Error: Cannot find module '@tracespace/config/webpack'".

Thanks
Pinal

mcous, 2020-04-13:
As details in the [Contributing Guide](https://github.com/tracespace/tracespace#development-setup), you must run `yarn install` before you are able to able to run `yarn start`. The `install` step installs development dependencies and links local packages (like `@tracespace/config`) using [yarn workspaces](https://classic.yarnpkg.com/en/docs/workspaces/). Did you run `yarn install`?

Like I said in my earlier posts, I will be able to respond faster and resolve your questions more easily if you please use [Gitter](https://gitter.im/tracespace/Lobby) rather than GitHub issues for support questions

pinal3257, 2020-04-14:
Hi Team

Yes, i installed yarn. after that i run "yarn start" command.
Can you please take a look on the attached file?

[afteryarnstart.txt](https://github.com/tracespace/tracespace/files/4472826/afteryarnstart.txt)

Thanks
Pinal

mcous, 2020-04-14:
@pinal3257 I Googled the error in your logs and found this issue: https://github.com/babel/babel/issues/11216. It appears to be a [bug in Node v13](https://github.com/nodejs/node/issues/32107). 

For best results, please ensure you are using an LTS version of Node (i.e. v10 or v12)