tracespace/tracespace #327: How does one convert one gerber file to one SVG via CLI?

andygock, 2020-04-28:
Can you use this tool to convert one gerber file to one SVG via CLI?

I've tried with no success. On a Windows machine, I keep receiving response `Error: No valid Gerber or drill files found` when issuing command `tracespace -B --force sample.gbr`. I get the same error regardless of whether `sample.gbr` exists or not.

Am I missing someone simple here?

What's an example command to convert a single gerber to SVG?

Thanks.

mcous, 2020-04-30:
Well this is embarrassing! The command you posted is correct (and the `--force` isn't necessary), except for the fact that input file path resolution is completely broken on Windows.

Fix incoming

closed 2020-04-30 by mcous

mcous, 2020-04-30:
Fix released with v4.2.1

PR #328 (2020-04-30): fix(cli): fix cli file path resolution on Windows
The input file resolution for the CLI was a little too clever, and attempted to resolve everything to absolute paths while replacing `~`s with the home directory and then passing everything into `globby` to also get any glob patterns. Unfortunately, [`globby` does not support Windows file paths](https://github.com/sindresorhus/globby#api), which meant that on Windows, any input file would be resolved to an absolute Windows path, which `globby` would fail to match with any files.

This PR:

- Gets rid of the explicit resolve to `process.cwd` bec [... cut]
commit 2eb1dc7b (2020-04-30): fix(cli): resolve file paths correctly on Windows (#328)

Fixes #327