tracespace/tracespace #334: germberTosvg

sangameshBappanna, 2020-06-23:
**Using this, it returns an object, how to get SVG out of that object.**

<script src="https://unpkg.com/gerber-to-svg@^4.0.0/dist/gerber-to-svg.min.js"></script>
<script>
  // global variable gerberToSvg now available
  var converter = gerberToSvg(input)
</script>

mcous, 2020-06-23:
Please see the [API documentation for gerber-to-svg](https://github.com/tracespace/tracespace/blob/main/packages/gerber-to-svg/API.md#gerber-to-svg-api) and [this example](https://flems.io/#0=N4IgzgpgNhDGAuEAmIBcICWA7JEAeAdABbwC2UIANCAGYYxhoDaoWAhqRGpjvsWRWqwA9lkRjuAHgCEAEQDyAYQAqATQAKAUQAEJcgD4AOlkl6oRrNu2SARsKQBPC1auTsABwCu8bfAfuIAF5DWnoIEO0MJGCQAHMIACcbRIBaD28Q5xdJYW8vHyiYsAA3WJTc+HzMyQB6Cvys1zBYBIx3HzAE2BiSSrBUGprPLHcAa1iCEVIa+KTU+GEUktiAAQA9ABYCAAYdmqQMMHgZxOSElIWl0oJSbAIAKzBqmubW9sbrV7b4D6sRLCOkRG3m0gW0SGEsE8nDEBHi8E0MBh8AAQg4AJJIAAUIVmZzSwPgIQAlMYXC5-oD6iCwRCoci4RAEUiIGI0ZiceBSuU8hkQKTLOSyeSgfkCGwkEhNMVWfAADKHcSJLEAclgRDYWHiKso2ggMrEoP02mAwpF2kpPgSEAliVB2iwEAA7toAGJhABKNtwCSxAvNVjNIuttoS4sl0tlCqOrOVKqgwglOu0fqNJqD5rxiWUwgAyqUsSGfQRrWBPFB4LqsYkEsIErrlsS06bBQGXBgaCma3Wm9b4J4EpZKcIYARu77x-62y5qfACNhHQkABLKACycvtywzIoAvlPzXvjNurEXEiXvQBBMDKfDwLHpOd0BhMbYAXX3IofBGKbCgngg9oqiq257iKQa1F87zCrUdiOBYtRmPoVDgNAcDwBgoiMOgABMACsqAAOwgDur7UOwnDcFm5yXMsep4Bw7gwMRQA)

```html
<!DOCTYPE html>
<html>
  <body>
    <input type="file" id="gerber-input">
    <output id="svg-output"></output>
    <script src="https://unpkg.com/gerber-to-svg@^4.0.0/dist/gerber-to-svg.min.js"></script>
    <script>
      const input = document.getElementById("gerber-input")
      const output = document.getElementById("svg-output")
     
      input.addEventListener('change', event => {
        const reader = new FileReader()
        
        reader.addEventListener('load', () => {
          gerberToSvg(reader.result, (error, svg) => {
            if (error) return console.error(error)
            output.innerHTML = svg
          })
        })

        reader.readAsText(input.files[0])
        input.value = ''
      })     
    </script>
  </body>
</html>
```

closed 2020-06-23 by mcous

sangameshBappanna, 2020-06-24:
Thank you, That was help full. Tracespace is a wonderful application.

sangameshBappanna, 2020-06-29:
> Please see the [API documentation for gerber-to-svg](https://github.com/tracespace/tracespace/blob/main/packages/gerber-to-svg/API.md#gerber-to-svg-api) and [this example](https://flems.io/#0=N4IgzgpgNhDGAuEAmIBcICWA7JEAeAdABbwC2UIANCAGYYxhoDaoWAhqRGpjvsWRWqwA9lkRjuAHgCEAEQDyAYQAqATQAKAUQAEJcgD4AOlkl6oRrNu2SARsKQBPC1auTsABwCu8bfAfuIAF5DWnoIEO0MJGCQAHMIACcbRIBaD28Q5xdJYW8vHyiYsAA3WJTc+HzMyQB6Cvys1zBYBIx3HzAE2BiSSrBUGprPLHcAa1iCEVIa+KTU+GEUktiAAQA9ABYCAAYdmqQMMHgZxOSElIWl0oJSbAIAKzBqmubW9sbrV7b4D6sRLCOkRG3m0gW0SGEsE8nDEBHi8E0MBh8AAQg4AJJIAAUIVmZzSwPgIQAlMYXC5-oD6iCwRCoci4RAEUiIGI0ZiceBSuU8hkQKTLOSyeSgfkCGwkEhNMVWfAADKHcSJLEAclgRDYWHiKso2ggMrEoP02mAwpF2kpPgSEAliVB2iwEAA7toAGJhABKNtwCSxAvNVjNIuttoS4sl0tlCqOrOVKqgwglOu0fqNJqD5rxiWUwgAyqUsSGfQRrWBPFB4LqsYkEsIErrlsS06bBQGXBgaCma3Wm9b4J4EpZKcIYARu77x-62y5qfACNhHQkABLKACycvtywzIoAvlPzXvjNurEXEiXvQBBMDKfDwLHpOd0BhMbYAXX3IofBGKbCgngg9oqiq257iKQa1F87zCrUdiOBYtRmPoVDgNAcDwBgoiMOgABMACsqAAOwgDur7UOwnDcFm5yXMsep4Bw7gwMRQA)
> 
> ```
> <!DOCTYPE html>
> <html>
>   <body>
>     <input type="file" id="gerber-input">
>     <output id="svg-output"></output>
>     <script src="https://unpkg.com/gerber-to-svg@^4.0.0/dist/gerber-to-svg.min.js"></script>
>     <script>
>       const input = document.getElementById("gerber-input")
>       const output = document.getElementById("svg-output")
>      
>       input.addEventListener('change', event => {
>         const reader = new FileReader()
>         
>         reader.addEventListener('load', () => {
>           gerberToSvg(reader.result, (error, svg) => {
>             if (error) return console.error(error)
>             output.innerHTML = svg
>           })
>         })
> 
>         reader.readAsText(input.files[0])
>         input.value = ''
>       })     
>     </script>
>   </body>
> </html>
> ```

can you help me pass options to gerbertosvg function? I am using javascript as mentioned above.

mcous, 2020-06-29:
@sangameshBappanna as written in the documentation, `gerberToSvg` takes an options object as its second argument

```js
gerberToSvg(gerberFileContents, optionsObject, doneCallback)
```

The options available are described here: https://github.com/tracespace/tracespace/blob/main/packages/gerber-to-svg/API.md#options