tracespace/tracespace #345: Aperture macros troubles ??

mr-miky, 2020-10-11:
I'm experiencing anomalies in the display of the gerber generated by kicad version 5.99 or 6.0.0 RC.
It is still a development version but incompatibilities have emerged when macros are used.
What happens is that the rounded pads of the smd components look wrong. I have checked the pcbs several times on https://gerber.ucamco.com/ and they appear correct.
I have already reported the anomaly to those who follow the development of kicad and have inserted an option to return to the "old method" for drawing pads.
I wanted to ask you if it is possible to make your engine handle these gerber as well because there are several pcb makers that use it as an online preview and I personally use https://tracespace.io/view/ because I can't trust the gerber of the RC version of kicad (it has already happened to me that 2 pcb makers rejected the gerber even if they were perfectly displayed on gerber.ucamco.com
It is enclosing a test zip. Look at the resistor pads 0603, the soldermask gives an idea of ​​the profile it should have but copper is a dot. The size of the pcb when it is loaded is wrong.

new mode
[Rev1.0_new.zip](https://github.com/tracespace/tracespace/files/5360713/Rev1.0_new.zip)

old mode
[Rev1.0_old.zip](https://github.com/tracespace/tracespace/files/5360716/Rev1.0_old.zip)

mcous, 2020-10-12:
Thanks for the report! It looks like we have a pair of bugs spread across `gerber-parser` and `gerber-plotter` that are causing this aperture macro to fail to render. I should be able to get a fix up tonight thanks to your very helpful reproduction files

closed 2020-10-13 by mcous

mcous, 2020-10-13:
The fix is now live as v4.2.3. Thanks for your help!

PR #346 (2020-10-13): fix: accept negative tool mods and macro expression args for outline primities
This PR fixes #345. There were three separate bugs that combined to produce the bad output reported:

- The parser did not know how to handle negative tool modifiers in `%AD` tool definitions
- The parser, for outline primitives in particular, did not allow macro expressions to be used as parameters to the primitive
- The plotter did not know how to plot an outline primitive using macro expressions as its parameters
commit 09d4a289 (2020-10-13): fix(parser,plotter): resolve issues with macro parsing and plotting (#346)

- Fix regex issue that caused parser to ignore negative tool modifiers in %AD tool definitions
- Allow parser to output macro expression functions as outline primitive arguments
- Ensure plotter can handle an outline primitive using macro expressions as its parameters

Fixes #345