tracespace/tracespace #348: Doesn't fully handle Gerber X2 files and drill files

baryluk, 2020-11-09:
I discovered on tracespace.io , pcbway and pcbgogo websites that the Gerber X2 files (both layers and drill ones), are not fully supported, some layers are not recognized.

[image]

Attaching an example.
[pic-programer-gerber-x2.zip](https://github.com/tracespace/tracespace/files/5508365/pic-programer-gerber-x2.zip)

The files are from KiCad 4.x, and from KiCad master branch.

For example top layer is not recognized despite the header containing proper information:

```
%TF.FileFunction,Copper,L1,Top*%
%TF.FilePolarity,Positive*%
```

Similarly the plated vias, are not recognized, but the file header has information:

```
%TF.FileFunction,Plated,1,2,PTH,Drill*%
%TF.FilePolarity,Positive*%
```

mcous, 2020-11-09:
Thanks for the report @baryluk! You're right, the parser does not recognize the `%TF` commands. They're definitely on my mind, though, so I'll consider this the official feature tracking ticket.

For now, does setting the file type manually in the board settings cause everything to render properly?

baryluk, 2020-11-15:
If I manually use the Protel-style extensions on files, it looks mostly correct, with exception of some drills probably, because I have two drill files (PTH and NPTH).

So, I believe it is mostly recognition of file function from the header that is missing. I noticed the code only check at file names and extensions, for guessing, but with X2 extensions will all be `gbr`, the filenames shouldn't be relayed on (i.e. different EDA software and people, use different naming conventions for top and bottom layer for example), and instead the header info should be used.

From what I understand about Gerber, the `%TF.FileFunction` and `%TF.FilePoliarity%`, must be in the header, so probably not a whole file need to be scanned (to check for them). Scanning whole file would be fine, because even with million lines, it would be really fast thing to do, but maybe there is some other line that indicates end of header or start of normal data (both in Gerber 1 and Gerber X2), so it can be made even faster. For the start `^TF.FileFunction((,[^,]+)+)\*%$` maybe.

I you are really busy, I can take a look at implementing it. The guessing function already receives the filename, so I guess it can retrieve the content of the file too. My guess is it is already in the browser memory.

I also noticed, that if I check the all layers in the viewer, the soldermask file, is recognized as soldermask (probably from the KiCad style filename), but has wrong polarity in the viewer.

```
%TF.FileFunction,Soldermask,Top*%
%TF.FilePolarity,Negative*%
```

[image]

The soldermask should be reversed here. Not sure where where the viewer should get the bounding boxes for the negation, but it is probably somewhere, because in some other gerber viewer software it did work for me.