tracespace/tracespace #352: Why is Trailing zero mode used as a default in Excellon parser?

thetroy, 2020-12-11:
I have an older excellon document that suggests Leading is the default. Wondering if there's a specific reason T was chosen as the default.

https://github.com/tracespace/tracespace/blob/d1a1417b98a4994988f1af2ce88aa7cf5037c0c7/packages/gerber-parser/lib/_parse-drill.js#L166-L169

We'd been forcing the use of L due to a bug, on our App. For backward compatibility, I'd hoped to add a 'backupZeroSuppression' option to the parser (similar to the plotter). That is, unless there's a reason not to do that. Hence the question: Why is Trailing zero mode used as a default in Excellon parser?

mcous, 2020-12-12:
I suspect we have been looking at [the same Excellon "spec"](https://web.archive.org/web/20071030075236/http://www.excellon.com/manuals/program.htm). The answers to this question aren't particularly satisfying, unfortunately. That old, archived, not-actually-a-specification says:

> The CNC-7 uses leading zeros unless you specify otherwise through a part program or the console

The key detail here is that drill files are the opposite of Gerber files: in Gerber, you specify which zeros to drop, but in NC drill files, you specify which zeros to _keep_. So, the closest thing to an Excellon spec that exists says "keep leading zeros and drop trailing zeros by default".

On top of that:

- Anecdotally, at the time I selected the default, I was running into a lot of drill files from Eagle and PADS that implicitely used "keep leading zero / suppress trailing zeros" without specifying it
- Changing the default would be a breaking change

In `v5`, I'm looking at the following changes, with nothing in particular set in stone

- Coordinate strings are not decoded in the parser
- When a batch of parsed layers hit the plotter, to determine the zero suppression settings of a single layer, it:
    1. Checks the layer file itself
        1. First checks if there's a zero suppression command
        2. If not, examines all coordinate strings to see if any start or end with zeros
    2. If that didn't work, use the most common setting from the other files in the batch
    3. If that doesn't work, fall back to leading suppression (keep trailing) because that's how numbers work in the real world

Uh yeah, so TL;DR: 🤷. I don't really like it as a default, it will not continue to be the default, and if it's a problem for you now I'd accept a PR to add a `backupZero` option to the parser given I have no timeline at all on v5.

thetroy, 2020-12-12:
ah yes. I remember tripping over this excellon-keep/gerber-suppress inconsistency long ago. I'm going to double-check what's happening with this file again in light of the inconsistency. If that goes as expected I'll proceed with adding `backupZero`. I'll probably end up using it to implement ii from my app (I use parser and plotter directly). 

i-ii-iii sounds great! If you have a task list for v5 I might pick some off. Even simple chores like converting files to TS. No worries, if you choose not to.

closed 2020-12-12 by thetroy