tracespace/tracespace #353: G93 code in drill file rendered as drilled hole

adneedle, 2020-12-15:
I am using the pcb-stackup and gerber-to-svg libraries to render a composite image of a bare board. The layout tool I use, Mentor Xpedition, seems to add a G93 code to its drill output to indicate the relative zero coordinate (`G93X0Y0`). _parse-drill.js ignores the G93 part of the line and recognizes the X0Y0 as a drill coordinate. It then seems to incorrectly drill the last defined tool at 0,0.

For example, this drill file below will render an extra T10 0.463in hole at 0,0. I think there should probably be a check for G93 codes at the start of the line so they can be ignored as drill coordinates.

```
; Format  : 2.4 / Absolute / Inch / Trailing
; Contents: Thru / Drill / Non-Plated
M48
INCH,LZ
FMAT,2
ICI,OFF
T07C0.0660
T08C0.1020
T09C0.1280
T11C0.3290
T10C0.4630
%
M72
G90
G93X0Y0
T07
X035Y0229
T08
X0161Y007
X00735Y0151
T09
X0445Y0
X0
X0445Y0225
X0
T11
X0046Y006
T10
X0297Y00605
Y01645
T00
M30
```

mcous, 2020-12-15:
Thanks for the report. Would accept a PR to ensure G93's are no-ops rather than drill hits

mcous, 2023-01-15:
I believe the offending regex is here:

https://github.com/tracespace/tracespace/blob/49c9a22addccc70c1d1a830a9225328a91942080/packages/gerber-parser/lib/_parse-drill.js#L21

I don't recall why the regex does not start with `^`, but it probably should, and I think that would fix the issue

Gianarci, 2024-02-20:
I've the same issue with tracespace 5.0.0-alpha. How to fix it?