tracespace/tracespace #354: Spaces around equals in gerber pad shape variable definition formula

adneedle, 2020-12-15:
When Mentor Xpedition creates shape definitions in gerber files, it seems to add spaces around equal signs in the variable definition formulas. For example:
```
%AMVB_Custom_Oblong*
21,1,$2-$3,$3,0,0,$1*
$2 = $2/2*
$3 = $3/2*
$6 = $2-$3*
1,1,$3X2,$4X$6,$5X$6*
$6 = 0-$6*
1,1,$3X2,$4X$6,$5X$6*
%
```

This causes the regular expression in _parse-macro-block.js, `var RE_VAR_DEF = /^(\$[\d+])=(.+)/`, to fail to match. The pad variable definitions are then out of sync for the file and it tries to use undefined variables. 

I am not sure if the spaces around the equal signs comply with the gerber spec, but I ended up locally changing the regex to `var RE_VAR_DEF = /^(\$[\d+])\s?=\s?(.+)/` to make it work in my situation.

mcous, 2020-12-15:
Very interesting. This is 100% not compliant with the Gerber File Format specification 

> **3.3 Character Set**
> ...
> Space characters can only be used inside strings (see 3.6.5). 

If possible, I would definitely raise a bug report with Mentor for this.

From a tracespace perspective, yeah I think allowing the equal sign regex to have some optional spaces makes sense for leniency, **if that leniency is common in other Gerber viewing tools**. The risk in leniency is that it may give someone a false sense of security that their invalid Gerber file will be understood by a fab. I would also probably limit the regex to just literal space characters rather than `\s`, but I'd accept a PR for this functionality.

adneedle, 2020-12-16:
Thanks. It does not surprise me very much that the Xpedition gerber output is buggy. I am not sure if we have seen a fab issue before, but that may be because we almost always use ODB++ now for the fab data.