tracespace/tracespace #360: Fails to load .gdo gerber files

mehulmshah, 2021-04-29:
Tried to upload a set of Gerber Files exported from Mentor PADS (extension .gdo).

Error I get is: `Uncaught TypeError: Cannot read property 'type' of undefined` from `_pad-shape.js:239`

Those few lines of code:
```
var runMacro = function(mods, blocks) {
  var emptyMacro = {shape: [], box: boundingBox.new()}
  var exposure = 1

  blocks = blocks || []

  return blocks.reduce(function(result, block) {
    var shapeAndBox

    if (block.type !== 'variable' && block.type !== 'comment') {
```
Line 239 is the last line^, looks like `block` is undefined... not sure if there needs to be some type of parsing adding for Mentor files, or if there just needs to be a check that returns from the function if `block===undefined`

kasbah, 2021-04-29:
Can you provide the file? If not, can you patch the code? What happens when you add a check in there?

```js
var runMacro = function(mods, blocks) {
  var emptyMacro = {shape: [], box: boundingBox.new()}
  var exposure = 1

  blocks = blocks || []

  return blocks.reduce(function(result, block) {
    var shapeAndBox

    // add this below
    if (block == null) {
      return result
    }

    if (block.type !== 'variable' && block.type !== 'comment') {
    
```

mehulmshah, 2021-04-29:
I'm not sure, I'm using the `pcb-stackup` library which abstracts out all of this code so I can't add the check in my own code. But I would imagine that it may work if that check is present?

I can't provide the whole gerber, but essentially here is the issue:
Most gerbers have the basic options in the beginning, and then the actual instructions, like so:
```%FSLAX24Y24*%
%MOIN*%
...
%ADD10C,0.00500*%
%ADD13C,0.00600*%
%ADD18C,0.00700*%
%ADD16C,0.00800*%
%ADD12C,0.01000*%
```

This set from Mentor had those options, and then defined some custom shapes, and then finally had the instructions:
```
%MOIN*%
%FSLAX24Y24*%
...
%AMVB_RECTANGLE*
21,1,$1,$2,0,0,$3*
%
%AMVB_Custom_COMPLEX 0.61X0.3 A0.183 S1_Mirrored*
$4 = $2X-0.0091*
$5 = $3X0.0000*
$6 = $2X0.0000*
...
%ADD10C,0.00500*%
%ADD13C,0.00600*%
%ADD17C,0.00700*%
...
```
If I remove the custom shape definition sections (AMVB_RECTANGLE, AMVB_Custom..., etc) from the file, then it parses correctly with no issues. There may be some rectangles that are drawn incorrectly, as those shapes are referenced in the instructions later in the file, but as far as I'm aware the gerbers are essentially correct for 99% of use cases.

mehulmshah, 2021-04-29:
@kasbah Adding that check prevents the error and allows the gerbers to render. Now the issue is that the gerbers are rendered slightly incorrectly (but this is at least better than the code breaking!)

Results of testing w/ if statement:
- Using original Gerber file, gerber loads but copper layer is incorrect (changing gerber settings does not help)
- Deleting custom shape definitions from top of Gerber file, gerber loads but copper layer is incorrect (same as above)
- Replacing everything before actual %ADD... *% instructions with a random Altium Gerber file that I know loads correctly, the Gerber loads completely perfect! I don't want to share the files publicly but I can explain in more detail if anyone wants to talk via a more private medium (discord, zoom, etc.)

mcous, 2021-04-29:
Best guess is that [`parseMacroBlock`](https://github.com/tracespace/tracespace/blob/4dd28dfa5a99512460469ce61430e4803ec434cf/packages/gerber-parser/lib/_parse-macro-block.js#L9) is not recognizing a primitive code in one of these aperture macros, hitting the end of its `if/else` ladder and implicitly returning `undefined`. This garbage data then gets fed into the plotter, resulting in this throw.

@mehulmshah are you able to post the full contents of these aperture blocks? Everything (including newlines) starting with `%AM` and ending with `%`. In your post above,  you've cut some off with `...` but the entire contents of the macro would be helpful to track down why the parser didn't recognize a given block.

You could also trying running the file through just the parser using the example code:

```js
var fs = require('fs')
var gerberParser = require('gerber-parser')

var parser = gerberParser()

parser.on('warning', function(w) {
  console.warn('warning at line ' + w.line + ': ' + w.message)
})

fs.createReadStream('/path/to/gerber/file.gbr')
  .pipe(parser)
  .on('data', function(obj) {
    console.log(JSON.stringify(obj))
  })
```

I'd be curious what, if any warnings popped out

mehulmshah, 2021-04-29:
@mcous I think your assessment is correct. I'd prefer not to post the contents publicly on this Issue, but if you let me know your email I'm happy to privately message you the full file so you can take a look. I'll try running it through the parser as well and see what the results are.

Thoughts on adding Interpolation settings to Gerber Settings? It looks like other CAM software is able to open these Gerbers (although I'm not sure how considering the Gerber says G74 (Disable circular interpolation), but it only renders correctly when I manually change to G75 (enable multiquadrant circular interpolation)

mcous, 2021-04-29:
I would say the arc thing is separate from the macro parsing issue, and that I'd heavily suspect you're running into #82 with those arcs. If so, then I don't think extra plotting settings are the right move to work around a bug in the plotter

mehulmshah, 2021-04-29:
Ah yeah you're correct, I think #82 is the more relevant one. I emailed you & @kasbah the files for this issue if you wanted to take a closer look.